ros-controls/ros_controllers #157: Make ForwardCommandController and ForwardJointGroupCommandController thread-safe

adolfo-rt, 2015-01-22:
There is currently no synchronization between the non-rt thread that executes the subscriber callback and the rt thread that runs the controller update method. A `realtime_tools::RealtimeBuffer` could be used to fix this, as is done in other controllers.

adolfo-rt, 2015-01-22:
@ipa-fxm you should be aware of this, as I believe you use these controllers.

ipa-fxm, 2015-01-23:
Thanks for the hint...
Although we haven't experienced any problem with the current version, I guess #160 fixes this issue.

ipa-fxm, 2015-02-06:
#161 fixes it for the forward_command_controllers
However, similar race conditions might occur in other controllers e.g. effort_controllers/JointPositionController as they are using a shared member [here](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/effort_controllers/include/effort_controllers/joint_position_controller.h#L167)