ros-controls/ros_controllers #162: diff_drive_controller reacts badly to dt < 0

paulbovbel, 2015-02-19:
As discussed [here](https://groups.google.com/forum/#!topic/ros-sig-robot-control/18VRoTEV1vM), the behaviour of diff_drive_controller when passed a -ve dt for update period is very undesireable.

This can be warned against in docs/tutorials, but perhaps some sort of corrective action could be done by the controller as well, such as overriding negative periods with zero, or entirely skipping the loop.

efernandez, 2015-02-20:
@bmagyar @adolfo-rt

bmagyar, 2015-02-22:
I'd prefer having a fix for it over doc warnings. Anyone working on it?
 
Anyone wants to? ... or shall we throw a dice? :)

adolfo-rt, 2015-02-24:
Let's first take a look to where the control period is used. Apparently, it's only used [here](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/diff_drive_controller/src/speed_limiter.cpp#L89-L90), in the speed limiter. Would it make sense to actually handle negative control periods?, like taking the absolute value of the period?. If it does makes sense, apart from the implementation changes, we could add a note to the respective [documentation](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/diff_drive_controller/include/diff_drive_controller/speed_limiter.h#L71).

Negative control periods is something that in general should not occur, and should be enforced at the robot hardware level. As stated in the [link](https://groups.google.com/forum/#!topic/ros-sig-robot-control/18VRoTEV1vM) of the OP, the actual policy on what to do in these circumstances should not be imposed by ros_control, though.

bmagyar, 2015-12-01:
A related one #202

bmagyar, 2016-01-29:
This issue has been hanging in the air for almost a year. Let's put at least one nail into the coffin of it. 
How about adding a check & warning & link to this issue? It wouldn't solve the problem but would help people not trip over it.

I can add these changes if you agree.