ros-controls/ros_controllers #164: Add wiki example

saznetro, 2015-02-22:
Please provide a configuration example for skid steer.

mikepurvis, 2015-05-26:
This is not a substitute for proper wiki docs, but Jackal is a full, working, open-source example of a skid steer robot:

https://github.com/jackal/jackal/blob/indigo-devel/jackal_control/config/control.yaml#L5
http://docs.ros.org/indigo/api/jackal_tutorials/html/simulation.html

bmagyar, 2015-05-27:
Thanks Mike!

adolfo-rt, 2015-05-27:
@saznetro, it seems that the know-how is out there, but not in the easy-to-find place you're expecting. How do you feel about adding a simple and self-contained example to the ROS wiki from what Clearpath has made available, so others like you can easily find it?. The wiki can be edited by anyone with an account, and we could proof-read your contribution.

saznetro, 2015-05-28:
@adolfo-rt Okay, I will do that.

adolfo-rt, 2015-05-29:
Your intentions are much appreciated. Ping us back when you have something to review.

saznetro, 2015-06-14:
I have finished it, please review it:
http://wiki.ros.org/diff_drive_controller#Skid_steer

bmagyar, 2015-06-14:
* Params are showing up again,
* 3 undocumented params documented
* Jackal example added

Nice :+1: !

adolfo-rt, 2015-06-15:
Good stuff!. @saznetro, thanks for putting the example together, and to @bmagyar for the review.

closed 2015-06-15 by adolfo-rt