ros-controls/ros_controllers #165: joint trajectory controller - account for brakes

sachinchitta, 2015-03-06:
With industrial robots, the first action that often happens before a first move (e.g. on power up or after an e-stop) is for the brakes to be switched off. This takes a finite amount of time and the actual trajectory then lags the desired trajectory. Exposing a general mechanism to account for this "startup delay" (which happens sporadically but in known situations) would be nice. This information should also get propagated up to other nodes (like MoveIt!). A proper way to fix this would be to expose the brakes through the ROS API so a higher level node could deal with the brakes before sending trajectories.

adolfo-rt, 2015-03-06:
Interesting question.

Title-wise, I think the scope of the question goes beyond the joint trajectory controller. I see the following points that would need to be addressed:

Expose brakes as a hardware interface. This would make sense if there were controllers to actually toggle the brakes, or at least report its state to the outside world (a `BrakeState` message).

The other issue is taking into account the finite time it takes to enable power. One solution would be to simply not update controllers while in e-stop or powering up, maybe even have them stopped (so commands would be rejected). The downside of this is that stopping _all_ controllers doesn't allow you to monitor state, which is undesired if you're remotely debugging an e-stopped robot.

A better alternative that would require some work, would be to allow the controller manager to update a subset of its controllers. I can imagine a mechanism that would allow a trigger to start/stop some controllers, maybe depending on their hardware interface, something like 'stop all controllers claiming resources from anything other than `JointStateInterface`'. This would be robot-specific, but a general mechanism for exposing this behavior could exist. This way, one could keep on reporting state (joints, brake states, IMUs, whatever) when power (for drives, not electronics) is disabled.

sachinchitta, 2015-09-09:
I implemented this outside the joint trajectory controller so I am closing this. I do think its useful to know which controllers monitor state so they can still be updated when the system is e-stopped. I will open a different issue for that though.

closed 2015-09-09 by sachinchitta