ros-controls/ros_controllers #167: diff_drive_controller assumes joint parent == origin

T045T, 2015-03-27:
I'm currently working on a URDF file for a differential drive robot that's build from several (potentially reusable) parts. This means that the drive wheel joints' ```parent``` element is not always ```base_link```, or even in a different position at all.

In [diff_drive_controller.cpp](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/diff_drive_controller/src/diff_drive_controller.cpp#L440), ```left_wheel_joint->parent_to_joint_origin_transform``` is reported as the distance to origin. However, that is not true.

In my specific case, the parent link is in the same location, but oriented differently, resulting in a wheel separation of 0.

I have two ideas on how to address this (not counting a change to the URDF):
* Look up the wheel separation via ```tf```
* add a ```wheel_separation``` parameter that overrides the lookup in the URDF

Please let me know what you think of these - I can implement either or both of them, but would like some feedback first.

bmagyar, 2015-03-27:
Hi,

First of all, which ROS distro are you using? Not that it makes any difference, but it is easier to get new stuff released into indigo.

Actually I have already implemented a solution for this using a parameter for 'wheel_separation' and also one for 'wheel_diameter' (on a robocup project for this year we reinvented the wheel in a way where cylinders are not good approximations anymore).

So the solution for this is done. I can prepare a pull request of it from my private repo, but I will need a test. Do you mind implementing it? I can give you access to my fork so you can push there directly. There are plenty of examples on tests in the appropriate folder.

As a final note, I find it hard and risky to add a tf listener to a realtime-safe module.

T045T, 2015-03-27:
I'm using indigo - sorry I didn't mention that before.

Sure, I can write a test for this :)

And you're right, I forgot about the realtime requirement - although in this case it would just be a one-time lookup of a (hopefully) static transform.

bmagyar, 2015-03-29:
#169

bmagyar, 2015-06-14:
@adolfo-rt  I believe we can close this one.

closed 2015-06-15 by adolfo-rt