ros-controls/ros_controllers #173: joint_trajectory_controller does not publish feedback

mkoval, 2015-04-14:
The `FollowJointTrajectory` action can periodically return feedback about the actual state, desired state, and controller error. It would be nice if the `joint_trajectory_controller` published these messages.

This message is almost identical to the `control_msgs/JointTrajectoryControllerState` message that is already being published by the controller. I am willing to implement this feature myself and send a pull request, but I need some guidance on how to make this real-time safe.

I assume that I can't directly call `ServerGoalHandle::publishFeedback` because it is not real-time safe. It is also not wrapped in `RealtimeServerGoalHandle`. The most obvious answer is to add a `publishFeedback` method to `RealtimeServerGoalHandle::publishFeedback` and call it from `updateState` in the controller Is there any reason why this function is not  currently implemented?

adolfo-rt, 2015-04-15:
[Related post](https://groups.google.com/forum/#!topic/ros-sig-robot-control/pYAdBNIgsYs). Thanks for making it explicit as an issue.

I'll try to take a look at how to make the implementation in the coming days. In the meantime, you can take a look at how goal messages are published through the [RealtimeServerGoalHandle](https://github.com/ros-controls/realtime_tools/blob/indigo-devel/include/realtime_tools/realtime_server_goal_handle.h).

dcconner, 2015-04-20:
Note I'm also looking into ServerGoalHandle with #174.

One useful distinction with the published state that I might suggest is to make the feedback tied to changes in waypoint segments.  That way we are not just giving redundant information.

mkoval, 2015-04-20:
I've been swamped, so I haven't had a chance to look at `ServerGoalHandle` much.

> One useful distinction with the published state that I might suggest is to make the feedback tied to changes in waypoint segments. That way we are not just giving redundant information.

I would strongly prefer to get feedback at `state_publish_rate`. Waypoints are just a mechanism for specifying a `q(t)` spline. It would be undesirable to get different feedback messages from two trajectories that happen to have different parameterizations, but are otherwise identical.

It is frustrating that `joint_trajectory_controller` implements an `actionlib` interface, but I have to subscribe to an unrelated topic to get feedback. This also makes it difficult to associate feedback with a particular `JointTrajectoryControllerState` message with an actionlib goal.

miguelprada, 2016-04-28:
Has anyone actually made any progress on this?

I have locally implemented some (potentially non-realtime-safe) changes to RealtimeServerGoalHandle and to JointTrajectoryController, which I would be glad to prepare a couple of PRs with and see what you guys think.

bmagyar, 2016-04-28:
Show us what you've got!
Please create your PR-s against `kinetic-devel` (I've just added it, you may need to pull first)

miguelprada, 2016-04-28:
There. These are actually my first two PRs ever, so I apology in advance for any formal mistakes I surely have made. :wink:

bmagyar, 2016-04-29:
Thanks! I'm traveling at the moment but will try to test and give feedback
on Sunday!

alextoind, 2016-05-18:
I've tried the proposal today, but it seems to behave just the same as before (don't know why). Maybe I have some other problems.

miguelprada, 2016-05-19:
Nice to hear that, @alextoind!

bmagyar, 2016-05-22:
Hi guys,

I've merged #219 and https://github.com/ros-controls/realtime_tools/pull/18. Wanted to push it through before the Kinetic release on Monday so we're not breaking API or ABI on a non-beta distro :)

bmagyar, 2016-05-24:
Gentleman, I think we can call this done.
Closing...

closed 2016-05-24 by bmagyar