ros-controls/ros_controllers #175: diff_drive_controller forces URDF inconsistency with real robot

rgariepy, 2015-04-23:
If the URDF is to be accurately representative of a robot, the left motor will have its axis directed to the left and the right motor will have its axis directed to the right. However, this conflicts with diff_drive_controller, as it assumes that publishing positive velocity to both wheels will drive the robot forward and vice versa. Making the Gazebo mechanicalReduction tag negative will not help here, as ros_control doesn't know about it and it's meant for feedback calculation anyway.

Ideally, the URDF would be accurate *and* the joint output calculated would also be accurate, such that the hardware interface itself doesn't have to have any consideration of which side the motors are mounted on.

There is a rabbit hole here which results in a fully general controller, but I'm wondering about the opinions of either introducing a scaling factor *or* a flag which can tick-tock us towards the convention of forwards motion = right motor moving in the negative sense.

bmagyar, 2015-04-28:
Hello Ryan,

It is not standard where one implements the directional control of one's motor as it can be in the controller board, a man-in-the-middle communication layer between ros_control and the controller board, or the diff drive controller as you are suggesting. 

I see your point, and we indeed already have multipliers in the internal version of the controller in PAL. I believe we could push these changes upstream to relieve the controller from such constraints.

adolfo-rt, 2015-05-04:
> If the URDF is to be accurately representative of a robot, the left motor will have its axis directed to the left and the right motor will have its axis directed to the right.

Not on all robots. Some will have their joint axes parallel, and some anti-parallel. Your robots seem to fit in the latter case, ours in the former. Both alternatives are valid, and as long as there is no REP or similar document proposing a best practice, no alternative can be favored.

Best practices aside, the multiplicative factors that @bmagyar mentions would do the trick. They were originally implemented to take into account divergence between a wheel's nominal and actual size, but negative values could do the trick. If you can think of a better alternative, we can discuss it as well.

bmagyar, 2016-04-17:
Feel free to re-open if you wish to continue this discussion.

closed 2016-04-17 by bmagyar