ros-controls/ros_controllers #182: Partial joint trajectory controller

beatrizleon, 2015-07-08:
We are planning to work on modifying the joint_trajectory_controller to send a partial trajectory to our robots, so it will be possible to send a trajectory for only a subset of joints.

Any advise on how to implement it?

adolfo-rt, 2015-07-08:
There are two things to take into consideration as context:
 1. Trajectory commands can have a starting time (`header.stamp` field of [`trajectory_msgs/JointTrajectory`](http://docs.ros.org/api/trajectory_msgs/html/msg/JointTrajectory.html)) set to now (the default) or at any time in the future.

 2. When a new command arrives, the new trajectory is [merged](http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement) with the current one. That is:
  * The new trajectory does *not* necessarily substitute the current one.
  * The merging process occurs at the moment a new command arrives.

The two above points raise the following question: What will be the policy for assigning values to unspecified joints?.

Assigning the current state, or the current command to unspecified joints could be an alternative for trajectories starting now (`header.stamp == 0`), but is definitely a bad idea for trajectories starting in the future. This is because these unspecified joints might have moved elsewhere between now and the actual start time.

If we go down this road, we could allow not specifying joints for trajectories starting now, but not for trajectories starting in the future. This could OTOH lead to confusion as to why it works in some cases (start now) and not in others (start later).

If we want to support unspecified joints for all trajectory start times (now and later), then a more in-depth analysis of the implementation should be done. I don't think this will be straightforward.

beatrizleon, 2015-07-09:
Would it be posible for trajectories starting in the future to set the unspecified joint state as the current target? So not the state when the trajectory is received but the one when the trajectory actually starts moving?

adolfo-rt, 2015-07-09:
That would likely require a rewrite of the controller logic.

In the current implementation, the new trajectory is merged with the current one as soon as it's received. It is not merged when the time has come for it to start.

The rationale behind this is that we do [batch-merging](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L500) (merge the entire new trajectory into the current one). This process is non real-time safe, as it [dynamically allocates resources](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h#L352), and takes a potentially unbounded time (new trajectory can be arbitrarily long), so we do it in the non real-time ROS API callbacks.

It should be possible to achieve what you propose by deferring trajectory merging to the real-time code path, so at every cycle the controller would determine which trajectory segment to use. I'm also confident that this can be done in a real-time safe way. Surely there are many details to sort out, but the idea sounds technically feasible. The only caveat I see is the amount of work this task might require. If you're willing to implement it, I'm willing to review it. The controller has a pretty good test suite, so we should be able to detect most regressions due to a major refactor.

beatrizleon, 2015-07-09:
One question: As the merge of the trajectory is happening per joint, wouldn't it be possible to not touch the joints that are not specified (so leave the current red planned trajectory as in the graphs) and only merge the ones that are specified?

We are thinking of using this for a five finger hand, so for example send a specific grasp as the target trajectory, but if one of the fingers detect a contact, we want to send a new trajectoru for it to stop it, and allow the other fingers to continue to their goal.

adolfo-rt, 2015-07-09:
> As the merge of the trajectory is happening per joint, wouldn't it be possible to not touch the joints that are not specified (so leave the current red planned trajectory as in the graphs) and only merge the ones that are specified?

That sounds reasonable. two things to get right are:
* When using the action interface, trajectory segments are associated to a specific goal handle. If certain joint segments are not touched, we need to make sure that the goal management logic remains valid (maybe there's nothing to do, I just haven't checked).

* This solution strategy might still require some effort, as the whole implementation assumes that after some point, all segment specifications have the expected joint order and number of joints. There will be some different or extra name remapping to do.

> We are thinking of using this for a five finger hand, so for example send a specific grasp as the target trajectory, but if one of the fingers detect a contact, we want to send a new trajectoru for it to stop it, and allow the other fingers to continue to their goal.

We do that as well, and we've found it simpler and less invasive to do this reasoning outside the controller. Subscribing to the `state` topic of the controller, you have actual and desired positions, velocities and accelerations (only desired for acc) of the controller joints, which is enough to prepare a new controller goal. This is valid as long as you don't need to do this at real-time frequencies (which we don't).

beatrizleon, 2015-07-15:
We discussed the different alternatives but we still think it would be better if this is handled in the controler side. I was looking at the joint trajectory controller code and I think the problem is the way a trajectory is defined. If I understood correctly, at any given time, there is a current_trajectory that is being executed (or held if there is no other goal send).

A trajectory is defined as a set of segments. Each segment is defined as:
Segment:
* const ros::Time& traj_start_time,
* const trajectory_msgs::JointTrajectoryPoint& start_point 
* const trajectory_msgs::JointTrajectoryPoint& end_point 
* const std::vector<unsigned int>& permutation (Permutation vector mapping the expected joint order to the message joint order)
* const std::vector<Scalar>& position_offset (offsets due to wrapping joints to zero)

A trajectory point is defined as:
* positions()
* velocities()
* accelerations()
* time_from_start()

So each segment is defined as a start and end point vectors of positions for all joints. So, as far as I can see, the problem is that this segments are not created per joint but as a vector of all joints at a given time. I am not seeing clearly how (with the current implementation) we can create an additional segment for only a partial set of joints without creating it for all of them. 

Any ideas?

adolfo-rt, 2015-07-15:
> So each segment is defined as a start and end point vectors of positions for all joints. So, as far as I can see, the problem is that these segments are not created per joint but as a vector of all joints at a given time. I am not seeing clearly how (with the current implementation) we can create an additional segment for only a partial set of joints without creating it for all of them. 

Yes, that's why. You nailed it. If you wanted to leave certain joints unspecified for a trajectory that starts in the future, you would have to defer trajectory merging to the real-time loop. I expect many things will need to be re-thought to accommodate this usecase.

The less invasive alternative is to only support partial specifications for trajectories starting now, but I'm not fully convinced that it's a good idea.

beatrizleon, 2015-07-15:
When a new trajectory goal arrives with partial joints, couldn't we sample the current trajectory for each future point in time of the new trajectory and use that to fill the positions for the partial joints that are not specified?

So filling the unspecified joints with the interpolation of where that joint is meant to be in the future. I think that would take into account the time when the trajectory is planned to start.

That it make sense?

adolfo-rt, 2015-07-15:
What you propose will work as long as the trajectory state at the future start time remains unchanged (pos, vel, acc). The following is an example in which proves the point:

* Controller gets a new command `c1` starting at time `t_future`, where `foo_joint` is unspecified.

* We sample the current trajectory at time `t_future`, and assign that value to `foo_joint` from time _future`, and for the entire duration of the new command.

* A new command `c2` arrives starting now, and ending before `t_future`, which leaves `foo_joint` in a different position than the one we had sampled.

* We reach `t_future`, and `foo_joint` jumps from where `c2` left it to where it though it would be at this moment.

beatrizleon, 2015-07-20:
yes. Also I was trying to quickly implement it to see how it will work. I managed to interpolate the new trajectory joints but I don't think this is the right approach. We will have to interpolate for each time of the new trajectory of partial joints which can have times that are different from the current trajectory which will create problems of sorting the times and interpolating in the new and old times. Additionally the computed curve is different for the old segment compared with the two curves with an interpolated point. 

I was thinking that the correct way to do it should be to change the current implementation to allow independent control of each joint. So allow to store segments of different time per joint. Do you think we can modify the current implementation to enable this?

closed 2016-10-31 by bmagyar