ros-controls/ros_controllers #188: [rqt_joint_trajectory_controller] Save/restore plugin settings

adolfo-rt, 2015-08-19:
Remember selected controller manager and controller from last session, and restore them if they are available. This will allow to save some serious mouse clicking.

ipa-mdl, 2015-08-19:
I guess you just want to start controllers, but not stop them?

adolfo-rt, 2015-08-19:
> I guess you just want to start controllers, but not stop them?

Not starting or stopping the controllers themselves. That can be done by `rqt_controller_manager`. 

Currently, if you have a controller in your `rqt_joint_trajectory_controller` plugin, like:
[image]

and you`Ctrl+C`, then relaunch `rqt`, what you get is this:
[image]

The proposed behavior restores the controller you had last time, if it's running (or nothing if it's no longer running), so you go back to the first image, with less clicking around.

Restoring the state works also if you have multiple instances of the plugin, so you can now save rqt perspectives where one instance controls say, the right arm, another the head, another a hand.

ipa-mdl, 2015-08-19:
Ah, I got confused..
+1 for rqt_joint_trajectory_controller restore :D

commit 360fee39 (2015-08-25): rqt_jtc: Save and restore plugin settings

Fixes #188.

- Save current controller_manager and controller selection on plugin close
- Restore last selection if controller manager and controller are running
closed 2015-08-25 by adolfo-rt