ros-controls/ros_controllers #197: [imu_sensor_controller] Covariance updates use wrong index

tappan-at-git, 2015-10-02:
The covariance update loops (starting [here](https://github.com/ros-controls/ros_controllers/blob/indigo-devel/imu_sensor_controller/src/imu_sensor_controller.cpp#L99)) update the element i (the number of IMU sensors/realtime publishers) of each covariance matrix once for each element of the matrix.

    void ImuSensorController::update(const ros::Time& time, const ros::Duration& /*period*/)
    {
      using namespace hardware_interface;
    
      // limit rate of publishing
      /* outer loop over the realtime publishers*/
      for (unsigned i=0; i<realtime_pubs_.size(); i++){
        if (publish_rate_ > 0.0 && last_publish_times_[i] + ros::Duration(1.0/publish_rate_) < time){
          // try to publish
          if (realtime_pubs_[i]->trylock()){
            // ...
            // Orientation covariance
            if (sensors_[i].getOrientationCovariance())
            {
              // inner loop over the elements of the covariance
              for (unsigned j=0; j<realtime_pubs_[i]->msg_.orientation_covariance.size(); ++j){
                // j is not used in the inner loop
                realtime_pubs_[i]->msg_.orientation_covariance[i] = sensors_[i].getOrientationCovariance()[i];
              }
            }
            // ...
            realtime_pubs_[i]->unlockAndPublish();
          }
        }
      }
    }

adolfo-rt, 2015-10-05:
A pull request against `indigo-devel` would be much appreciated.

tappan-at-git, 2015-10-05:
Done.

closed 2015-10-05 by adolfo-rt