ros-controls/ros_controllers #202: Why is  diff drive controller giving me weired commands?

cyborg-x1, 2015-11-30:
For me in a standard example the diffdrive controller always counts down.

On one axis command counts down with around -0.2 per second and the other one with around -2.0 per second.

It starts doing that without publishing anything to cmd_vel and publishing to cmd_vel does not change anything on this behaviour. Btw I am using the full setup (without having URDF) from the example.
http://wiki.ros.org/diff_drive_controller#Complete_description

Is there something known, I might do wrong?

bmagyar, 2015-11-30:
What do you mean by counting down? turning backwards? accelerating backwards? 
Is this a real robot or a simulated one? 

Do you have position feedback from the wheels? Cannot be that the position there is very noisy?

cyborg-x1, 2015-11-30:
Example just after loading the controller: (1 column / per second)
Command Wheel1: -0.2,  -0.4,  -0.6,  -0.8
Command Wheel2:   -2,      -4,     -6,     -8

I forget to mention currently I do not have a realtime kernel (if that could be the problem).
The position in the beginning is zero, the topic cmd_vel is not being published at all. 

Position is always zero when not on a real robot, or even with motor values, there is no change.  Also publishing cmd_vel does not do a single effect. In my opinion the wheels shouldn't start turning at all in the beginning right (if everything is zero (also all positions))? 

I created a little running example: https://github.com/cyborg-x1/ros_control_test/tree/diffDrvProb just put it in a workspace and compile (maybe rename folder to control_test) and run the launch file. Or just look at the code first, maybe you already see a mistake there...

cyborg-x1, 2015-12-01:
I found the issue ... I copied it from that question from here, because I was in need of a example for that ...
http://answers.ros.org/question/202320/ros_control-difficulty-getting-custom-hardware-interface-to-run/

well here is the mistake:
ros::Duration d = ts - ros::Time::now() ; //Its negative ... !

must be like:
ros::Duration d = ros::Time::now() - ts;

Then it works...

closed 2015-12-01 by cyborg-x1

bmagyar, 2015-12-01:
Nice catch! Passing negative time would cause most controllers to have some issues probably.

cyborg-x1, 2015-12-01:
Yeah, also how it came into my mind ... I just woke up, saw this line of code in my mind and I was like ... oh of course ... there it is ...  - sometimes you just need to sleep over it :smile: