ros-controls/ros_controllers #204: diff_drive_controller is hard-coded into the 'odom' reference frame

tappan-at-git, 2015-12-13:
There are a few cases where it is valuable to publish odometry (msgs and tf) in a frame other than the standard odom. In general, it would be better for frame_ids to not be hardcoded.

bmagyar, 2015-12-14:
Hi,

In defense of the current configuration, I'd like to reference this REP:
http://www.ros.org/reps/rep-0105.html

Could you elaborate on your specific case please?

tappan-at-git, 2015-12-14:
No problem,

My specific application involves multiple robots operating off of the same master, publishing to /tf using essentially namespaced /tf frames. (i.e. robot1 publishes a transform from /robot1/base_link to /robot1/odom). Previous implementations used remapped /tf topics which allowed for frame collisions, but publishing to the same topic has allowed for much cleaner transfer of information.

As tf_prefix [is deprecated](http://wiki.ros.org/tf2/Migration#Removal_of_support_for_tf_prefix), multi-robot applications under one master need to be able to specify alternate frames than those specified in REP 105.

It's frustrating to go against REP, but so far it's been the cleanest solution.

bmagyar, 2015-12-14:
Sounds reasonable. Let's head over to the PR.

bmagyar, 2015-12-15:
As the related PR is merged, so is this issue solved. 
Feel free to reference it in future for new REPs or other issues. Multi-robot systems are becoming more common nowadays.

closed 2015-12-15 by bmagyar