ros-controls/ros_controllers #208: Family of X_controllers/JointGroupYController using PIDs

bmagyar, 2016-01-29:
Greetings fellow ros-controllers,

I was wondering if there is interest in having grouped controllers that plug to type _X_ joint interfaces while accepting commands for type _Y_ joint interfaces.

These would be useful for robots that are homogenic in their joint interfaces (Effort, for instance) but need to be controlled in some other manner (eg: Position, to keep straight, or Velocity to keep a fixed speed) in cases where the hardware abstraction layers cannot be fiddled with.

Last week I took `effort_controllers/JointPositionController` and combined it with `forward_command_controller/ForwardJointGroupCommandController`. My `effort_controllers/JointGroupPositionController` can be found here:
https://github.com/bmagyar/ros_controllers/blob/position-effortgroupcontroller/effort_controllers/src/joint_group_position_controller.cpp

My current implementation is nowhere pretty nor generic, but it works. For example it gets the Valkyrie do these funky moves:
https://www.youtube.com/watch?v=tLCpJvqgtRQ

Your input on this would be much appreciated, thanks!

bmagyar, 2016-02-09:
What do you guys think? @ipa-mdl @davetcoleman

ipa-mdl, 2016-02-09:
I don't think that we at IPA have the need for these controllers at the moment (@ipa-fxm: ?), but it might be useful for others.
Most of your code could be moved into a common template.
In addition https://github.com/ros-controls/ros_controllers/pull/187 might help.

IMHO the limit enforcing should be done by the RobotHW implementation.

Perhaps it would make sense to implement a chaining mechanism that would add PID controllers to all controllers.
I am going to try this with joint limits filtering. (https://github.com/ipa320/cob_control/issues/72)

davetcoleman, 2016-02-09:
I thought our current controllers already convert, say, position commands to, say, effort commands? What does this group controller feature do differently?

ipa-mdl, 2016-02-09:
The forward (group) controllers do not convert, they pass all data directly.

bmagyar, 2016-02-12:
The only thing this controller adds compared to `effort_controllers/JointPositionController` is grouping. 

For further improvements I can remove the joint limit enforcing and I was also thinking about making it templated over the "Command" and "Output" types.