ros-controls/ros_controllers #210: Provide a ListControllers service like ros-control does

swhart115, 2016-03-31:
ros-control allows you to make a service call to get the active controllers.  

The name is: "/<robot_name>/controller_manager/list_controllers"
and the type is controller_manager_msgs::ListControllers 

It would be helpful for the ros-controllers controller manager to provide a similar functionality.  The only way i can see to get active controllers without this is looking at what the MoveIt! controllers.yaml file has (loaded on the param server).   But if you're not using MoveIt!, you're kind of out of luck....

bmagyar, 2016-04-01:
Hi,

In a `ros_control` compliant system there is usually one controller manager per robot. The `ros_control` controllers (whether they are hosted at `ros_controllers` or not) are managed through the _controller_manager_ (eg: load, start, stop, unload).

The MoveIt! controller yaml file is generated by the MoveIt! wizard. Such a file is not an accurate representation of controller states on an active robot (controllers can be stopped/started/swapped/removed any time based on some high level behaviours for instance).

I'm pretty sure I didn't get your question, but I hope these lines are in the ball-park of a satisfactory answer :) I'd need more details on your specific use-case.

swhart115, 2016-04-01:
Yes, there seems to be some confusion.  Forget what i said about the MoveIt! yaml; you are right that it is only something that tells MoveIt! about the possible controllers, not the active/running controllers.

But this is sort of my point.  The ros_control controller_manager has a service call that lets you get the active controllers, the ros_controllers controller_manager does not have such a service call.  This ticket is to request that one be added for "consistency."  Other then parsing the topic and parameter names i'm not sure how to get this information programmatically. 

I am writing an application where i would like to have access to this information easily, and wish to support ros_control and ros_controllers in as a consistent way as possible.

bmagyar, 2016-04-01:
I think we are getting there.
Are you sure this is the right repo and not this one? --> https://github.com/fetchrobotics/robot_controllers

swhart115, 2016-04-01:
Yes, it looks like you are correct. the way-too-similar naming convention isnt confusing at all! thanks

bmagyar, 2016-04-01:
:+1: 
The way I like to look at it, is that both `ros_control` and `robot_controllers` are generalisations of the PR2 controller framework but they went different lengths in slightly different ways.

Closing this issue.

closed 2016-04-01 by bmagyar