ros-controls/ros_controllers #215: Accessing Controllers Through Code

Jmeyer1292, 2016-05-05:
Greetings, I apologize for asking a question without thoroughly reading the code but I'm rather pressed for time.

My scenario is this:
I'm running a controller on a three-DOF robot using `velocity_controllers::JointPositionController`. I've based my work somewhat on @davetcoleman's boilerplate. I control these through the exposed ROS topic `<controller_name>/command`. I load the controllers through the `controller_manager` helper nodes.

My problem:
I would like to have direct access to the controllers in code. I want to be able to, for example, call `setCommand(pos, vel)` instead of the position only interface exposed through the ROS level interface.

Is there a reasonable way to do this? (I saw `ControllerLoaderInterface` but wasn't sure if there was a more direct route).

Thank you.

bmagyar, 2016-05-05:
Using mixed interfaces has not been explored in the official repos of `ros_control` though I heard about custom solutions. You could do something similar:
* (if you are using urdf) add 2 interfaces to your joint
* (if not using urdf) add your joint to 2 interfaces of your robot hardware
* implement a custom version of that controller that takes 2 interfaces per joint name you configure it with

I hope my answer is in the ballpark of your issue.

Controllers are used through a plugin system, they are not exposed to programmatic changes, only to operations through the `controller_manager` and their own interfaces.

bmagyar, 2016-08-31:
Since there wasn't any activity on this for a while now, I'm closing the issue.
Feel free to reopen if you have something to add.

closed 2016-08-31 by bmagyar