ros-controls/ros_controllers #216: Create separate GitHub repositories for each package

alainsanguinetti, 2016-05-09:
Hello,

I am trying to use the joint_trajectory_controller in my project. This means that I have a fully set-up catkin workspace where I have my whole project, which is kind of a "ros release" that I want to deploy to several machines and that I track with git.
However, there is a problem in joint_trajectory_controller_impl.h, namely the definition of non-static non-inline functions in a header file, that I would like to fix and hopefully merge into the ROS official packages. 
My problem is, I don't know how I can simply clone that particular package, fix it and issue a pull request without having to pull the entire ros_controllers repository into a separate catkin folder, catkin_make install everything, switch workspaces, etc
I suggest that every package is a repository,(because every package is independent right ?) and that ros_controllers be the repository of the metapackage only. 

Would that make sense ?

bmagyar, 2016-05-09:
Hi,
I'm not sure I get your use-case, so help me out a bit more by starting with the usual things:
ROS version, operating system & version, way you set things up, error messages.
By the way:
If you need to use the joint_trajectory_controller, why don't you just install the binary package?
What's the exact issue with compiling controllers? I have done custom versions of a few at times and had no issues.
Carrying catkin-workspaces around can be done by using rosinstall files.

alainsanguinetti, 2016-05-10:
Hello, Thanks for your help and sorry for the poor quality of the PR, I'll do better next time!

The use case for the bug I am trying to fix is very simple: 
system is ROS Indigo on Ubuntu 14.04.4 64bits
the problem happens in this configuration:
- code1.h imports joint_trajectory_controller.h, definitions in code1.cpp
- code2.h imports code1.h, definitions in code2.cpp

when code1 and code2 get compiled to code1.o and code2.o, everything is OK, but link edition fails because the functions defined in joint_trajectory_controller_impl.h without inline or static keyword have been defined in different .o 

---
For the rationale behind providing packages as different repositories:

I would love to include in my project joint_trajectory_controller as a git submodule to be able of modifying it while still keeping track of improvements from the official codebase.
Also, I think it makes more sense to have one repository for every independently released package because since they are shipped separately, it should be possible to modify one independently of the other packages. 
Then, I don't see what does this repository map to in a catkin_workspace : it is not catkin_workspace(the source are not in a src/ folder) nor a package (not the structure of a package) so it is very difficult to work on it from an existing setup.
At last, some of the ROS github organizations have repositories for every packages while some have repositories for every "metapackages" ? It would be better to have one way of organizing the GitHub repositories.
Is it clearer like this ?

alainsanguinetti, 2016-05-10:
After working on it today, I realized that my designed was flawed : I was trying to use the controller as a separate object, not as a plugin.
I reverted to using it as a plugin so the need for fixing the multiple definition bug disappeared.
However the two issues raised above still apply according to me.

bmagyar, 2016-05-10:
Good. Also, stuff that is _impl is a header only implementation, not meant as a software library for people to include it in their project. The purpose of that package is the plugin. Actually I'm not even sure that the includes should be exposed, probably we could hide them...

`ros_controllers` is a metapackage that pulls in the rest of the controller packages. It is true that the repository is getting big, but there are not that many changes happening there since the codebase is quite stable so there wasn't a reason to pull them into other repos. Why would you want to work on the joint trajectory controller so that you need to have it on many computers? It is a very polished piece of software with plenty of abstraction. Not easy to jump in and do changes. Probably you are better off working on something from scratch.

What was the other issue?

alainsanguinetti, 2016-05-10:
Hello, you answered the two questions: the first one about defining functions in a header, the other about why the repository is organized like this. However, I would like to highlight that the repository as such can not be called a metapackage because it is not even a catkin_package at all.
As you suggest, I will probably move to my own tailored solution to control my robot, so I will not bother you anymore ! ;)

closed 2016-05-10 by alainsanguinetti