ros-controls/ros_controllers #218: Joint Trajectory Controller: acceleration data for Joint Handle?

alextoind, 2016-05-19:
_There's no acceleration data available in a joint handle_ (cf. [joint_trajectory_controller_impl.h](https://github.com/ros-controls/ros_controllers/blob/jade-devel/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_controller_impl.h#L411)).

Is it at least scheduled or it does not worth the effort at the moment?

bmagyar, 2016-05-19:
Personally I support everything that has enough demand for it.
The JointHandle pretty much reflects [sensor_msgs/JointState](http://docs.ros.org/api/sensor_msgs/html/msg/JointState.html). API/ABI changes should be either done fast or in ROS L ...

alextoind, 2016-05-20:
Thank you for your quick response.
I'm sorry but at the moment I don't think to have time to understand completely what changes should be done to implement this feature.
Maybe you can close issue.

bmagyar, 2016-05-20:
I'm going to leave it open, just labeled it as enhancement, thanks for the note! :+1: