ros-controls/ros_controllers #220: Support for UR5?

Chunting, 2016-05-26:
Hi, 
Is this controller support for Universal Robot UR5, CB3.2?
or anyone has developed a controller on UR5?

Thanks,
Chunting

carlosjoserg, 2016-05-26:
Hi @Chunting, yes, I've moved an UR3/CB3.

However, your question is more appropriate for the ROS Industrial organization, this is the main repo: https://github.com/ros-industrial/universal_robot, and particularly the driver I'm using is https://github.com/ThomasTimm/ur_modern_driver (it requires some ros-indsutrial packages, and probably will be transfered to ros-industrial in the future, see https://github.com/ThomasTimm/ur_modern_driver/issues/18).

Best
Carlos

bmagyar, 2016-05-26:
Thanks Carlos for answering, I suppose I can close it.

Chunting, 2016-05-26:
Thank you very much, you two are so kind.
I have set up the ur_modern_driver, and it runs well...

Thanks,
Chunting

bmagyar, 2016-05-26:
Good luck with it :+1: 

Closing the issue :)

closed 2016-05-26 by bmagyar