ros-controls/ros_controllers #233: Cryptic error message from joint_trajectory_controller when sending trajectories that occur completely in the past

BrettHemes, 2016-08-31:
Originally brought up in [ros_industrial/kuka_experimental/#69](https://github.com/ros-industrial/kuka_experimental/issues/69), the joint_trajectory_controller reports a rather useless error when a trajectory is sent that occurs completely in the past.
(i.e., trajectory.header.stamp + trajectory.points.back().time_from_start  <  ros::Time::now())

The error reported from calling SimpleActionClient::sendGoal(...) is as follows:

> [ERROR] [1472046592.407557364]: Unexpected exception caught when initializing trajectory from ROS message data.

Seems to be from line 518 [here](http://docs.ros.org/indigo/api/joint_trajectory_controller/html/c++/joint__trajectory__controller__impl_8h_source.html#l00518).  After catching and reporting the error, the action server reports success and the robot remains motionless.

After some thought I guess this is technically correct behavior due to the fact that the goal trajectory is in essence complete, but a more informative error message (perhaps from a check to see if the trajectory occurs in the present) might save some confusion in the future.  Thoughts?

Additionally, there is a disclaimer in the documentation for [initJointTrajectory](http://docs.ros.org/hydro/api/joint_trajectory_controller/html/c++/namespacejoint__trajectory__controller.html#a8b55b5e436dc0e49ad024d2f3bd7568c) that states:

> This function (joint_trajectory_controller::initJointTrajectory) does not throw any exceptions by itself, but the segment constructor might. In such a case, this method should be wrapped inside a try block. 

... but it is a bit buried.

bmagyar, 2016-08-31:
Hi Brett,

Thanks for the feedback!

I've taken a quick look and couldn't find (upon first check) that the Segment constructor ( [here](http://docs.ros.org/indigo/api/joint_trajectory_controller/html/c++/joint__trajectory__segment_8h_source.html) ) would throw anything other than `std::invalid_argument` exceptions.

Could you find out where this exception comes from? One way would be to remove the `catch(...)` part and get a stacktrace with our good friend `gdb`.

The solution for this should be to add explicit handling for these cases, maybe the explicit check for the condition you wrote above.

@adolfo-rt any thoughts?

BrettHemes, 2016-08-31:
The backtrace isn't as straightforward as I expected... I have included it below:
```
#0  0x00007ffff61bfcc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff61c30d8 in __GI_abort () at abort.c:89
#2  0x00007ffff61b8b86 in __assert_fail_base (
    fmt=0x7ffff6309830 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7ffff793bb64 "!pthread_mutex_unlock(&m)", 
    file=file@entry=0x7ffff793ba48 "/usr/include/boost/thread/pthread/recursive_mutex.hpp", line=line@entry=115, 
    function=function@entry=0x7ffff793e6a0 <boost::recursive_mutex::unlock()::__PRETTY_FUNCTION__> "void boost::recursive_mutex::unlock()") at assert.c:92
#3  0x00007ffff61b8c32 in __GI___assert_fail (
    assertion=0x7ffff793bb64 "!pthread_mutex_unlock(&m)", 
    file=0x7ffff793ba48 "/usr/include/boost/thread/pthread/recursive_mutex.hpp", line=115, 
    function=0x7ffff793e6a0 <boost::recursive_mutex::unlock()::__PRETTY_FUNCTION__> "void boost::recursive_mutex::unlock()") at assert.c:101
#4  0x00007ffff7909f79 in boost::recursive_mutex::unlock (this=0x7fffe4037bd8)
    at /usr/include/boost/thread/pthread/recursive_mutex.hpp:115
#5  0x00007ffff790d767 in boost::unique_lock<boost::recursive_mutex>::~unique_lock (this=0x7fffed789550, __in_chrg=<optimized out>)
    at /usr/include/boost/thread/lock_types.hpp:331
#6  0x00007fffd7d7be36 in actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >::goalCallback (this=0x7fffe4037bd0, 
    goal=...)
    at /opt/ros/indigo/include/actionlib/server/action_server_base.h:255
#7  0x00007fffd7d97d5a in boost::_mfi::mf1<void, actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>::call<actionlib::ActionServer<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >*, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const> (this=0x7fffe4048848, 
    u=@0x7fffe4048858: 0x7fffe4037bd0, b1=...)
    at /usr/include/boost/bind/mem_fn_template.hpp:156
#8  0x00007fffd7d96a2a in boost::_mfi::mf1<void, actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>::operator()<actionlib::ActionServer<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >*> (this=0x7fffe4048848, 
    u=@0x7fffe4048858: 0x7fffe4037bd0, a1=...)
    at /usr/include/boost/bind/mem_fn_template.hpp:171
#9  0x00007fffd7d943e8 in boost::_bi::list2<boost::_bi::value<actionlib::ActionServer<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<control_msgs::FollowJointTrajectory---Type <return> to continue, or q <return> to quit---
ActionGoal_<std::allocator<void> > const> const&> > (this=0x7fffe4048858, 
    f=..., a=...) at /usr/include/boost/bind/bind.hpp:313
#10 0x00007fffd7d9141e in boost::_bi::bind_t<void, boost::_mfi::mf1<void, actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<actionlib::ActionServer<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >*>, boost::arg<1> > >::operator()<boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> > (this=0x7fffe4048848, 
    a1=...) at /usr/include/boost/bind/bind_template.hpp:47
#11 0x00007fffd7d8dd96 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, actionlib::ActionServerBase<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<actionlib::ActionServer<control_msgs::FollowJointTrajectoryAction_<std::allocator<void> > >*>, boost::arg<1> > >, void, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>::invoke (function_obj_ptr=..., a0=...)
    at /usr/include/boost/function/function_template.hpp:153
#12 0x00007fffd7d9826d in boost::function1<void, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&>::operator() (this=0x7fffe4048840, a0=...)
    at /usr/include/boost/function/function_template.hpp:767
#13 0x00007fffd7d96c9e in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<control_msgs::FollowJointTrajectoryActionGoal_<std::allocator<void> > const>) (function_obj_ptr=..., 
    a0=...) at /usr/include/boost/function/function_template.hpp:153
[... cut ...]
#18 0x00007ffff73adb73 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/indigo/lib/libroscpp.so
#19 0x00007ffff73f5f94 in ros::AsyncSpinnerImpl::threadFunc() ()
   from /opt/ros/indigo/lib/libroscpp.so
#20 0x00007ffff5d7ba4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#21 0x00007ffff6a70182 in start_thread (arg=0x7fffed78a700)
    at pthread_create.c:312
#22 0x00007ffff628347d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

bmagyar, 2016-08-31:
Ok great. This is although a bit more cryptic than the error message but at
least some information is there.

Could you share the exact way to reproduce this issue?
If you've got some code that'd be best.

I'll check if there are tests handling trajectories in the past.

bmagyar, 2016-08-31:
The test `ignoreOldActionTraj` from
https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/joint_trajectory_controller/test/joint_trajectory_controller_test.cpp
seems to be doing just that, right? The difference I can see here is that
this test send an old trajectory and checks that this won't interrupt the
current one. How does this relate to what you are experiencing?

BrettHemes, 2016-08-31:
I can work on making a full example case but to answer your question in the meantime, the error occurs when sending a (very) old trajectory with no other trajectories in progress.

Here is a snippet that will generate the error (assuming that my_trajectory is a good trajectory_msg::JointTrajectory):
```
my_trajectory.header.stamp = ros::Time::now() - ros::Duration(10000);
actionlib::SimpleActionClient<control_msgs::FollowJointTrajectoryAction> ac("joint_trajectory_action", true);
if (!ac.waitForServer(ros::Duration(2.0)))
{
  ROS_ERROR("Could not connect to action server");
  return false;
}
control_msgs::FollowJointTrajectoryGoal goal;
goal.trajectory = trajectory;
goal.goal_time_tolerance = ros::Duration(1.0);
ac.sendGoal(goal);
ac.waitForResult(goal.trajectory.points[goal.trajectory.points.size()-1].time_from_start + ros::Duration(10));
```

bmagyar, 2016-09-01:
Thanks @BrettHemes ! 
Please have a look at the pull request is referred to ( #236 ) where I get the same error as you. Now we'd need to figure out a fix for it... Let's move the discussion to over there! I can add you to the repo so you can commit on top of mine.

bmagyar, 2017-06-28:
Issue was fixed, test has been added. Closing...

closed 2017-06-28 by bmagyar