ros-controls/ros_controllers #234: joint_position_controller build error

nxdefiant, 2016-08-31:
joint_position_controller fails to build on Debian testing (gcc 6.1.1, boost 1.61) with the following error:

```
/opt/ros/ros_catkin_ws/src/ros_controllers/effort_controllers/src/joint_position_controller.cpp: In member function 'virtual bool effort_controllers::JointPositionController::init(hardware_interface::EffortJointInterface*, ros::NodeHandle&)':
/opt/ros/ros_catkin_ws/src/ros_controllers/effort_controllers/src/joint_position_controller.cpp:88:41: error: no match for 'operator=' (operand types are 'boost::shared_ptr<const urdf::Joint>' and 'urdf::JointConstSharedPtr {aka std::shared_ptr<const urdf::Joint>}')
   joint_urdf_ = urdf.getJoint(joint_name);
```

bmagyar, 2016-08-31:
Which ROS distribution?
I've seen a similar issues these days when I was goofing around building
unreleased stuff from source.

It seems urdf updated their *ConstSharedPtr to use `std::shared_ptr`
instead of `boost::shared_ptr` but it hasn't been released yet.

I think I can quickly fix this tomorrow morning, or if you already have the
fix please open a pull request now and I merge it tomorrow.

nxdefiant, 2016-08-31:
ros is kinetic-devel, have not attempted to fix it yet.

bmagyar, 2016-08-31:
Btw I see the path you are using in the logs. Try not to build things in
/opt because it may mess things up big time. Have your sources in your home
and use catkin install with a path set to opt.

bmagyar, 2016-09-01:
Hey, 
Give a shot to the pull request I referred below! This is a non-invasive solution that should even keep current code untouched, I refactored the affected parts to use the name macros.

nxdefiant, 2016-09-01:
It compiles.