ros-controls/ros_controllers #238: JointEffortController linker error

bhaskara, 2016-09-23:
I get a linker error when compiling code that includes `joint_effort_controller.h` and instantiates a `JointEffortController` using clang on OS X.  This seems to be because `JointEffortController` is a typedef for `ForwardCommandController<EffortJointInterface>`, which is abstract due to the lack of implementation of `starting`, and the definition of this function is in the source file `joint_effort_controller.cpp` rather than the header file `joint_effort_controller.h` as required for a method of a templated class.  Had to work around by copy-pasting the definition into my source file: https://github.com/bhaskara/baxter_simulator/commit/e03655d614ddfef342344ca24763e52210f76f77.

bmagyar, 2016-09-23:
Hi,
Would you mind testing the changes you proposed and submitting it as a pull request against kinetic-devel please?

bmagyar, 2017-06-28:
Closing due to lack of activity.
Feel free to reopen if the problem persists.

closed 2017-06-28 by bmagyar