ros-controls/ros_controllers #243: diff drive controllers not respecting limits

anderwm, 2016-11-30:
The controller output to each wheel I get is larger than the input twist message, and it's not respecting the limits I set in linear/x/max_velocity.  Obviously I am missing something, but I'm digging through the code and it all looks right. Any insight, or anyone seen this problem?

I see things like

> ros.diff_drive_controller.mobile_base_controller: Added values to command. Ang: -0, Lin: 0.261922, Stamp: 1480543793.787620630
ros.vdrt: update controllers
ros.vdrt: write controller outputs: 3.09517 and 3.09517
ros.diff_drive_controller.mobile_base_controller: Added values to command. Ang: -0, Lin: 0.261922, Stamp: 1480543793.888160284
ros.vdrt: update controllers
ros.vdrt: write controller outputs: 3.11812 and 3.11812
ros.vdrt: update controllers
ros.vdrt: write controller outputs: 3.11812 and 3.11812
ros.diff_drive_controller.mobile_base_controller: Added values to command. Ang: -0, Lin: 0.261922, Stamp: 1480543793.988044526
ros.vdrt: update controllers
ros.vdrt: write controller outputs: 3.11812 and 3.11812
ros.diff_drive_controller.mobile_base_controller: Added values to command. Ang: -0, Lin: 0.261922, Stamp: 1480543794.088889932

anderwm, 2016-12-01:
Ok, wheel radius gets applied after maximum velocities are enforced...not sure if this is desired or not but I will close it.

closed 2016-12-01 by anderwm

bmagyar, 2016-12-01:
Hello,

Limits are enforced on body velocities.
The wheel radius part is only mapping body velocity to wheels. 
Although we could, controllers typically won't deal with joint limits, those should be specified in the URDF and handled by the RobotHW implementation.

Hope that helps,
Bence