ros-controls/ros_controllers #251: seq is 0 in differential_drive controller

Timple, 2017-01-11:
If we run the differential_drive controller in open_loop mode, the seq in the headers stays 0. Although the timestamp in the header is correct.

In the code, I don't see the sequency in the header being set anywhere, is this a bug?

bmagyar, 2017-01-11:
I think this is just something none of us ever needed. If you have a need
for this, please feel free to provide a pull request implementing it.

Timple, 2017-01-12:
Well, it is in the header definition that it should be sequential. We have some tf problems with our robot and this sequence being zero is what stood out. 

It might not be the issue, but I will make a pull request nonetheless next week to fulfill the specification of the header.

bmagyar, 2017-01-12:
I cannot hurt, but keep in mind that this controller is being used on many
robots in production already, has tons of hours of test time. Seq is too
simple of a banana peel to slip on :)

Tf issues usually arise when time is out of sync, perhaps you have a multi
computer system where you'd need to periodically sync clocks?

ipa-mdl, 2017-03-22:
> We have some tf problems with our robot and this sequence being zero is what stood out.

I [doubt](https://github.com/ros/geometry2/search?q=seq) that tf uses the sequence number..

romainreignier, 2017-03-31:
It does not hurt to have it but it seems that `seq` is kind of deprecated:
https://discourse.ros.org/t/is-the-sequence-number-in-header-message-deprecated/1528
http://answers.ros.org/question/55126/why-does-ros-overwrite-my-sequence-number/?answer=55132#post-id-55132

bmagyar, 2017-03-31:
Yeah, I'd prefer to not add support for deprecated stuff so late. I'll close both this and the referenced PR. Thanks for bringing this up though!

closed 2017-03-31 by bmagyar