ros-controls/ros_controllers #252: Build error with diff_drive_controller and URDF pointer types

mryellow, 2017-01-27:
```
src/ros_controllers/diff_drive_controller/src/diff_drive_controller.cpp:61:36: error: ‘LinkConstSharedPtr’ in namespace ‘urdf’ does not name a type
 static bool isCylinder(const urdf::LinkConstSharedPtr& link)
```

https://github.com/ros/robot_model/issues/171

Could simply be `liburdfdom-dev` with Jessie is `0.3.0-1+b1` but buildfarms are working so not sure what the issue is exactly. Linking here in-case it's something in `ros_controllers`.

mryellow, 2017-01-27:
https://github.com/ros-controls/ros_controllers/pull/246/files

Seems related. That new header has backwards compatibility for what type is to be used.

bmagyar, 2017-01-27:
Yep. Source-based builds have been suffering from inconsistencies, and the urdf namespace being spread out among 2 packages (or more?) didn't ease the process. It should be all fine now if you use the newest of everything and you'll probably need the patch of #246 . I'm planning to merge that as soon as all travis builds are fine.

A new release of ros_control & ros_controllers is coming up soon as well.

bmagyar, 2017-03-25:
This should be ok by now. Please reopen if it is still broken.

closed 2017-03-25 by bmagyar