ros-controls/ros_controllers #254: joint_trajectory_controller position with velocity

anderwm, 2017-03-03:
I am switching some dynamixel code over from the joint_trajectory_controller in the dynamixel stack to ros_controls. When I run (ros_controls) joint_trajectory_controller over the position interface I only get a position command. Although the it is a position interface, I expected a velocity command to come out of joint_trajectory_controller to act as a max velocity for that position command. In that way the time of each segment would be controlled (this is how the one in the dynamixel stack works).

Is this functionality possible with ros_controls joint_trajectory_controller? Possibly I should be using the velocity interface, but in reality it is a position interface that I have(dynamixel has) with a velocity I can set while going to that position.

ipa-mdl, 2017-03-04:
I guess this is solved in https://github.com/ros-controls/ros_controllers/pull/83, please try to rebase and test it

bmagyar, 2017-03-04:
Great, thanks for rekindling this effort!
Where do you need it? I'd prefer to put all new stuff on Kinetic and forwards.

anderwm, 2017-03-04:
Sorry, does this mean it isn't in kinetic-devel or indigo-devel( I use both ATM)? I found the posvel joint handle in kinetic, do I still need to rebase #141 to test it?

ipa-mdl, 2017-03-04:
>  I'd prefer to put all new stuff on Kinetic and forwards.

The code was originally targeted at hydro ;)

> Sorry, does this mean it isn't in kinetic-devel or indigo-devel( I use both ATM)? I found the posvel joint handle in kinetic, do I still need to rebase #141 to test it?

The posvel inteface was merged into indigo (#181) and is available in kinetic as well.
If you want to use the `joint_trajectory_controller` with posvel(acc?) interface then you should checkout my branch and try it, I have already rebased it to the latest indigo state.

anderwm, 2017-03-06:
Assuming I am tracking in the right direction: I was planning to use the posvel hardware interface like

`            hardware_interface::PosVelJointHandle joint_handle_pos_vel(jnt_state_interface.getHandle(arm_joint_names_[joint_id]),
                                                                       &arm_joint_position_command_[joint_id],
                                                                  &arm_joint_velocity_command_[joint_id]);`

 but it looks like the joint_trajectory_controller expects a posvelacc interface. Does this mean I should use posvelacc interface and pass it some unused memory location for the acceleration command in order to test the joint_trajectory_controller?

ipa-mdl, 2017-03-06:
> Does this mean I should use posvelacc interface and pass it some unused memory location for the acceleration command in order to test the joint_trajectory_controller?

Either omit the the acceleration value or patch the controller to support a posvel interface as well (preferred).

anderwm, 2017-03-09:
Ok I tested the pos_vel_acc_controllers/JointTrajectoryController by not using the acceleration in my hardware interface. I originally tried to copy what you did to make a pos_vel_controllers/JointTrajectoryController but I must have missed something because that didn't work.

Anyway, If there was any doubt, it does indeed create position and velocities and I used it on real hardware.

 I extended the dynamixel stack with a multi joint controller that takes position and velocity commands on multiple joints. I then used that as a bridge between the dynamixel stack way of doing things and ros_control in my hardware_interface. Long story short anyone with a dynamixel arm could be using/testing this code if they wanted. In particular it is needed if you want to have dynamixel and other ros_control things in the same kinematic chain in moveit.

In reality, I'm not sure how you could even call something a "Trajectory" controller if it doesn't at least do position and velocity.  If you are only commanding position at times I'm not sure it is possible to actually control the trajectory (at least between each line segment).

bmagyar, 2017-03-09:
Can we expect a pull request of this?

Have you had a look at the combined robot hardware interface?  May be worth
exploring if you want to ship a dynamixel compatible RobotHW with the
stack.

anderwm, 2017-03-09:
> Can we expect a pull request of this?

I didn't actually change anything in ros_controls or ros_controllers.  I only tested the pos_vel_acc_controllers/JointTrajectoryController in #141 from @ipa-mdl .  The only thing I did that might be pull worthy would belong in the dynamixel stack, and I have a little bit of testing left to do on it first. If somebody can use this I think I can provide it.  The dynamixel stack already has it's own joint_trajectory_controller that isn't ros_controls friendly which has a follow_joint_trajectory action server to interact with moveit.  However if all your joints aren't dynamixel you would have to do something like I did or write another custom joint_trajectory_controller.

> Have you had a look at the combined robot hardware interface?

I have looked at it a little and I think it would help me. However, I can't fully wrap my mind around how it works yet and documentation is sparse.  Hopefully I'll find time to figure it out.

Zabot, 2017-05-15:
What ever became of this and #141/#83?

closed 2017-10-06 by graiola