ros-controls/ros_controllers #261: Don't wrap angle differences optionally in joint position controllers

lucasw, 2017-03-27:
I'd like to be able to command a continuous joint to for example 20 pi and see it revolve 10 times and then stop.   Currently the angle wrapping (calls to angles.h `shortest_angular_distance()`) prevents this from happening.  Would it make sense to create a new joint type that doesn't have this property, or add options to the existing controller types to support it?

Another feature to go with this would be to reset the position, so that if the controller gets many revolutions away from zero the angle can be wrapped back down to near zero but only when requested.

http://answers.ros.org/question/197805/are-jointpositioncontrollers-supposed-to-wrap-angles-on-continuous-joints/

bmagyar, 2017-03-27:
This sounds to me like something you'd rather want to do in velocity and
integrate over time.

The position interfaces assume you are interested in setting the robot to a
given pose regardless of the execution.  This 20 pi example (excellent one
btw ;)) is something where one mixes end goal &  assumptions on execution.

Please give us a common use case where this would be beneficial.

http://answers.ros.org/question/197805/are-jointpositioncontrollers-
supposed-to-wrap-angles-on-continuous-joints/

<https://github.com/ros-controls/ros_controllers/issues/261>, or mute the
thread

dcconner, 2017-03-28:
I left a comment on the  answers.ros link.
Isn't this the difference between Revolute and Continuous joints in the urdf.
I would think a ros_controller for a revolute joint would (should?) handle the multiple revolution condition, where a continuous joint would handle with default wrapping.  I think this should be a modeling decision.

lucasw, 2017-03-28:
I tried out > 2pi limits span with revolute (using rrbot joint2 in the latest gazebo_ros_demos) and it acted like this: a lower limit of -4 and upper limit of 4 acted like a 2pi - 4 to 4.0 range.
It is doing wrapping in an unexpected way.
The `set_point` still goes from -4 to 4 though.

It would be nice to have it behave like @dcconner suggests instead.  It is mechanically reasonable to have a rotary joint with hard stops or soft stops that provide more than 360 degrees of travel (but probably far less than 720).  A cable tray might allow 400 degrees of rotation.

I would still like to have a continuous version as well as revolute limits that allow >2pi travel (putting very large numbers into the limits and hoping they never are reached sounds ugly).  I made the answers post a while back and I think I just want a convenient way to have an actuator spin around a few times and then stop without having to pay attention to it after the initial command.  It could be a wheel, a winch, a gear, or something else like that.

A parallel change would be to have an option to have any rotary joint report unwrapped rotations regardless of type - if it has spun 10 times since startup then it would be possible to know that from the angle directly without constant monitoring to see that a wrap event has occurred.

scpeters, 2020-04-01:
I've submitted #477 which uses the `angles:: shortest_angular_distance_with_large_limits` function instead of `shortest_angular_distance_with_limits`. I think this will resolve this issue.

matthew-reynolds, 2020-04-16:
Resolved in #466 (Which is basically the same as #477 :smile: )

closed 2020-04-16 by matthew-reynolds