ros-controls/ros_controllers #279: Intermittent test failures due (too) tight tolerances

tfoote, 2017-06-10:
In working on #278 I was running the tests before finalizing and I ran into several tests that failed.  

I've collected some I observed in: https://gist.github.com/tfoote/e2b270148ed396fe9db47fac915ab8c3

It's things like this:
```
[diff_drive_controller.rosunit-diff_drive_wheel_radius_param_test/testForward][FAILURE]
/tmp/dd/src/ros_controllers/diff_drive_controller/test/diff_drive_test.cpp:61
The difference between sqrt(dx*dx + dy*dy) and 1.0 is 0.019000000000003348, which exceeds POSITION_TOLERANCE, where
sqrt(dx*dx + dy*dy) evaluates to 0.98099999999999665,
1.0 evaluates to 1, and
POSITION_TOLERANCE evaluates to 0.01.
```

Where the error is just outside the tolerance window. A repeated test does not hit the same issue. 

The outputs in the gist are collected from 4-5 different testing runs.

tfoote, 2017-06-10:
It's also appearing on travis: https://travis-ci.org/ros-controls/ros_controllers/jobs/241399196 on 1 out of 3 architectures

```
[diff_drive_controller.rosunit-diff_drive_wheel_radius_param_test/testForward][FAILURE]
/root/catkin_ws/src/ros_controllers/diff_drive_controller/test/diff_drive_test.cpp:72
The difference between fabs(new_odom.twist.twist.linear.x) and cmd_vel.linear.x is 0.87958656594200002, which exceeds EPS, where
fabs(new_odom.twist.twist.linear.x) evaluates to 0.979586565942,
cmd_vel.linear.x evaluates to 0.10000000000000001, and
EPS evaluates to 0.01.
--------------------------------------------------------------------------------
[diff_drive_controller.rosunit-diff_drive_wheel_radius_param_test/testTurn][passed]
[diff_drive_controller.rosunit-diff_drive_wheel_radius_param_test/testOdomFrame][passed]
SUMMARY
 * RESULT: FAIL
 * TESTS: 3
 * ERRORS: 0
 * FAILURES: 1
```

bmagyar, 2017-06-10:
I don't remember having issues with these tests running them locally. Timing issues have always been a problem on Travis. Even when we ran on a single architecture, tests would fail from time to time.
Have you had similar experience on Travis before?

tfoote, 2017-06-10:
Sometimes on shared hosts performance can be an issue. In general we don't recommend having system performance tests as part of the CI. They're really only valuable on a standardized benchmark platform. However, most of these failures do not appear to be performance related. In general unit tests should not care about system performance.

bmagyar, 2018-03-23:
#318 and #319 have greatly improved robustness on our tests. Publishing time from the test seems to work well.

closed 2018-03-23 by bmagyar