ros-controls/ros_controllers #281: build error | the setGains() function do not have a antiwindup argument

Booooooosh, 2017-06-20:
When building effort controllers (the joint_velocity controller), it seems that the call to the pid_controller.setGains(p, i,d, i_max, i_min, antiwindup) filed due to the corresponding function within the module that locates inside the control_toolbox did not have a argument antiwindup.

I am using the ROS Indigo and Ubuntu 14.04, does anyone face the same question?

Many thanks!

bmagyar, 2017-06-20:
You overlooked the branch you are compiling from. The PR you referenced is against ROS Kinetic. Get the `indigo-devel` and you'll be fine. In case you don't need to modify the controllers you could also just install everything with `apt-get install ros-indigo-ros-control*`

closed 2017-06-20 by bmagyar

Booooooosh, 2017-06-21:
That's true, sorry for the confusion.