ros-controls/ros_controllers #291: joint_trajectory_controller - Always dropping first trajectory point

agutenkunst, 2017-08-11:
If a trajectory has no specified start time (0) and time_from_start of the first point is zero the first point will always be dropped. 

Changing
https://github.com/ros-controls/ros_controllers/blob/3ee947a8a51f0a7594e54b2a79b95bc01d1591dc/joint_trajectory_controller/include/joint_trajectory_controller/joint_trajectory_msg_utils.h#L57
< to <= stops this behaviour.

Not sure if this would be the best fix.

Any comments?

bmagyar, 2017-08-11:
Yes as that'd leave no time for your arm to make it to the first point. Better to fill `time_from_start` for the first point too.

ipa-mdl, 2017-08-11:
`time_from_start == 0` just holds for the starting point of a trajectory, if any.
And this starting point should better match the actual position of the actuator ;)

agutenkunst, 2017-08-14:
Just noticed that using OMPL there will always be the warning that the first point has been dropped which at first sounds like something you would not want a controller todo. Knowing that at time_from_start = 0 state == current_state seems pretty implicit.

bmagyar, 2018-03-23:
Any update on this?

jschleicher, 2018-10-25:
I'm working on this during WMD18

PR #366 (2018-10-27): dont print warning about dropped first point
Since this is expected behaviour, if no start time is given in header of the message.

The warning about dropped points isn's useful, if the first point just contains
the expected start point. Dropping it is expected behaviour, thus degrading this
to a debug output message.

A subsequent feature could be to check the current position against the
parametrized tolerance of this point.

Closes #291
PR #367 (2018-10-29): dont print warning about dropped first point
Same as #366 for melodic. I've only tested this on kinetic; please consider to double-check it.

Closes #291
commit bba5e65e (2018-10-29): dont print warning about dropped first point, if it is expected behaviour

The warning about dropped points isn's useful, if the first point just contains
the expected start point. Dropping it is expected behaviour, thus degrading this
to a debug output message.

A subsequent feature could be to check the current position against the
parametrized tolerance of this point.

Closes #291
closed 2018-10-29 by graiola