ros-controls/ros_controllers #292: Tests execution is becoming somewhat erratic

bmagyar, 2017-08-14:
Our testing suite is becoming unstable...

Some tests fail on travis:
https://travis-ci.org/ros-controls/ros_controllers/builds

While on the ROS jenkins even basic things started failing:
http://build.ros.org/job/Kdev__ros_controllers__ubuntu_xenial_amd64/30/

Could someone have a look at this please? @ipa-mdl maybe? or @efernandez ?

efernandez, 2017-08-21:
Did something got merged recently? I'm not aware of any recent change.

bmagyar, 2017-08-21:
In the last pr I merged I added a test for the allow multiple publulishers
param.

ipa-mdl, 2017-09-27:
Digging into the logs and the multiple publisher PR, it seems like the odometry contains NaNs or an invalid quaternion.

The test is calculating the velocity itself, it would be more robust to just read it from the twist message.
However, the NaN issue should get fixed.

ipa-mdl, 2017-09-27:
Possible reason:
The NaN issue is [by design](https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/diff_drive_controller/test/diffbot.h#L101).
The tests fail whenever the test was launch so fast that one of the first odomtry messages is read, which contains NaN.

I think one publisher should get moved into the test and the velocity should get read from the twist.