ros-controls/ros_controllers #300: How to debug PATH_TOLERANCE_VIOLATED error

jacknlliu, 2017-10-24:
I'm debugging a issue about 
```
[ WARN] [1508834853.402216761, 3680.601000000]: Controller arm_controller failed with error code PATH_TOLERANCE_VIOLATED
[ WARN] [1508834853.402397314, 3680.602000000]: Controller handle arm_controller reports status ABORTED
[ INFO] [1508834853.701254652, 3680.896000000]: ABORTED: Solution found but controller failed during execution
```

Could someone give more details about how reflect the path tolerance config from the `controller/state`? Is the `error` in `controller/state` equal to `path tolerance`?

The configuration of controller is
```
arm_gazebo_controller:
  type: velocity_controllers/JointTrajectoryController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
  constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.5
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}

  gains:
      shoulder_pan_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
      shoulder_lift_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
      elbow_joint: {p: 1.2,  i: 0.0, i: 0.1, i_clamp: 1}
      wrist_1_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
      wrist_2_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
      wrist_3_joint: {p: 1.2,  i: 0.0, d: 0.1, i_clamp: 1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  125
  action_monitor_rate: 10
```

bmagyar, 2017-10-24:
This is an error from MoveIt. The easiest clue is the `PATH_TOLERANCE_VIOLATED` error code. 

Check here:
http://docs.ros.org/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#a3672b7f6c7a3f869c5a0458062117d98

ipa-mdl, 2017-10-24:
> This is an error from MoveIt.

I would say this is an error from the trajectory controller that was reported by MoveIt.

> Is the error in controller/state equal to path tolerance?

The path tolerance is the bound for the error. If it exceeds threshold the error gets reported (but the motion continues! #48).
I am not sure which path tolerance is used, as far as I could read it in the code the path tolerances of the action goal overwrite the `trajectory` settings in the  `constraints` parameter.

jacknlliu, 2017-10-24:
trajectory controller gives too less report, it's not easy to debug.

ipa-mdl, 2017-10-24:
> trajectory controller gives too less report, it's not easy to debug.

You can set the verbose flag to get more details (#273)

jacknlliu, 2017-10-25:
@ipa-mdl using this flag in my ros launch file, like this?
```xml
<node name="arm_gazebo_controller_spawner" pkg="controller_manager" type="controller_manager" args="verbose spawn arm_gazebo_controller" respawn="false" output="screen"/>
```
How to include the parameter `verbose` for user application?

jacknlliu, 2017-11-02:
anyone know the details about this process?

ipa-mdl, 2017-11-02:
> anyone know the details about this process?

You have to set it in the source code. It is just a debugging option and is not safe for real-time operation.

jacknlliu, 2017-11-02:
@ipa-mdl  I'm not sure, should I re-compile this repository?

ipa-mdl, 2017-11-02:
>  I'm not sure, should I re-compile this repository?

Clone, change verbose flag, compile.

jacknlliu, 2017-11-02:
@ipa-mdl Thanks for your attention and very detailed reply.

closed 2017-11-02 by jacknlliu

geweihgg, 2018-07-12:
Hi, @jacknlliu ,
I'm using velocity_controllers/JointTrajectoryController to control the UR5 robot in Gazebo, and encountering the same problem as you. Could you tell me how to solve it?
Thank you very very much~
GeWei

jacknlliu, 2018-07-12:
@geweihgg I think you have met a hard problem about gazebo and ros control. We have more discusses in https://github.com/ros-controls/ros_controllers/issues/303. 

And could you try some workarounds following [Controlling joints with velocity commands](https://github.com/JenniferBuehler/joint-control-pkgs/wiki/Controlling-joints-with-velocity-commands).

geweihgg, 2018-07-12:
@jacknlliu , But I want to plan the path with MoveIt, that is to say, a JointTrajectoryController should exist. And I wonder that whether PATH_TOLERANCE_VIOLATED error is caused by imprecise PID values. Have you found any solutions for controlling UR using velocity_controllers/JointTrajectoryController?
GeWei

jacknlliu, 2018-07-12:
@geweihgg There is a JointTrajectoryAction in [joint-control-pkgs](https://github.com/JenniferBuehler/joint-control-pkgs/blob/master/joint_trajectory_execution/config/JointTrajectoryParamsTemplate.yaml), and you can try to replace the action topics in moveit config files. Then it may work with moveit. You can follow the [jaco-arm-pkgs](https://github.com/JenniferBuehler/jaco-arm-pkgs/wiki/Installation), and this can be an example of how to use the new type position-velocity controller which included in joint-control-pkgs with moveit.

geweihgg, 2018-07-12:
@jacknlliu , Thank you for your patience, I'll try it.
GeWei

qboticslabs, 2018-10-11:
The Tolerance checking implemented in joint trajectory controller is mentioning in the following code.  

[https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/joint_trajectory_controller/include/joint_trajectory_controller/tolerances.h](url)