ros-controls/ros_controllers #302: gripper_action_controller incompatible with Moveit

rfn123, 2017-10-27:
Hi,
I've already opened an issue in Moveit (https://github.com/ros-planning/moveit/issues/676), as I was not sure where this problem belongs to.

### Description
 I recently wanted to implement a gripper_action_controller within Moveit, and after some trying I found out that it always throws the error: 
`Name and joints must be specifed for each controller`

Looking into the code from Moveit [here](https://github.com/ros-planning/moveit/blob/5c5fcd7804daf349310db0d28539010e23c51dc0/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp#L72) tells me that this error is thrown when either '`joints`' (plural) or '`name`' is not found in the yaml file for the specific controller.

However, as stated in https://github.com/ros-controls/ros_controllers/issues/155 and looking into the gripper_action_controller code [here](https://github.com/ros-controls/ros_controllers/blob/9f5eaef834cc61bac4992decd8baa93e5fb6372a/gripper_action_controller/include/gripper_action_controller/gripper_action_controller_impl.h#L152) , the gripper controller seems to be only working with a single-dof gripper, meaning it can only be started if the .yaml param is specified as '`joint`' in the singular form.

It would be nice if someone can fix this problem:)

bmagyar, 2017-10-27:
Congratulations, you've been volunteered! 

Could you try a patch of changing that field to `joints` and see how it works? Happy to review the pull request.

(Sorry, seems you just can't avoid me haha)

rfn123, 2017-10-27:
@bmagyar just did a quick test with changing to `joints`, and Moveit complains about the next problem:
` The list of joints for controller /gripper_controller is not specified as an array
`
Guess that the joint name (which is only one value) needs to be converted to a  XmlRpcValue array somehow..see Moveit code [here](https://github.com/ros-planning/moveit/blob/5c5fcd7804daf349310db0d28539010e23c51dc0/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp#L94)

bmagyar, 2017-10-27:
Then modify your yaml file and specify it as an array.  Then to further
adjust the gripper controller have a look at the diff_drive_controller for
an example.

rfn123, 2017-10-28:
So after digging some more into the source code I actually found that my problem was somewhere else. The action_ns of the controller in the Moveit yaml file should be **`gripper_cmd`** and not `gripper_action`. And to avoid the error in the issue description the correct config files would be:

Moveit yaml:
```
controller_list:
 - name: gripper_controller
   action_ns: gripper_cmd
   type: GripperCommand
   default: true
   joints: 
      - gripper_joint
```
and the ros_control yaml:
```
  gripper_controller:
    type: "position_controllers/GripperActionController"
    joint: gripper_joint
```

Would be nice though if the param is standardized or at least documented somewhere :)

closed 2017-10-28 by rfn123