ros-controls/ros_controllers #303: Add more controllers support

jacknlliu, 2017-11-03:
I got many ros controllers from [robot_mechanism_controllers](http://wiki.ros.org/robot_mechanism_controllers) , which including 
- [JT Cartesian Controller](http://wiki.ros.org/robot_mechanism_controllers/JTCartesian%20Controller)
- [Cartesian Wrench Controller](http://wiki.ros.org/robot_mechanism_controllers/CartesianWrenchController)
- [Cartesian Twist Controller](http://wiki.ros.org/robot_mechanism_controllers/CartesianTwistController)
- [Cartesian Pose Controller](http://wiki.ros.org/robot_mechanism_controllers/CartesianPoseController)

Could these support in ros controllers?

bmagyar, 2017-11-03:
Thanks for bringing this up. You may have noticed the overlap between these controllers and the ones in `ros_controllers`. The difference historical: the ones you've posted are from the `pr2_controller_manager` days which is where `ros_control` originates from. That system was specific for the PR2 robot, while `ros_control` is robot-agnostic.

I agree that we should have a look at those controllers and see if they are worth porting over.

ipa-mdl, 2017-11-03:
It might be even possible to wrap the PR2 controller interface to our controller interface, but I this is worth the effort only if the controllers are generic (as @bmagyar  already pointed out).

bmagyar, 2017-11-03:
I was actually thinking to propose to the PR2 folks to consider
reimplementing the `pr2_controller_manager`-related software with
`ros_control` while keeping interface support but I think it'd almost be
the same anyways.

graiola, 2017-11-05:
I think it should be possible to transform them in ros_controllers, but I have two concerns:
- Which robot we use to create the tests? These controllers use efforts as commands and we would need an effort-controlled robot to test them (perhaps rrbot?).
- Are these controllers guaranteed to be real-time safe?

bmagyar, 2017-11-06:
Well... It turns out we don't actually have tests for any of our _effort controllers_ and `rrbot` only has dummy inertial parameters set up but in theory that should still allow tests to work.

bmagyar, 2017-11-28:
@jacknlliu any thoughts on this?

jacknlliu, 2017-11-29:
@bmagyar A lot of times, I want to use the effort controller interface(including joint trajectory controller and cartesian space controller), but I searched many ROS packages, especially ROS-Industrial and the GitHub projects about ROS, I find that no works on this, especially work with gazebo. I think this is very important for modern robotics research in Gazebo and ROS.

And I try to use the joint trajectory controller with effort interface, but we should first make the gravity compensation for Gazebo simulation robot, or the robot will not stable only with PID configuration?  The same thing in joint velocity interface in Gazebo, see [Controlling joints with velocity commands](https://github.com/JenniferBuehler/joint-control-pkgs/wiki/Controlling-joints-with-velocity-commands).

bmagyar, 2017-11-29:
Gazebo has 2 approaches for this. One is through the `ros_control` interface where they leave  controller implementations up to the user and one where the controllers are provided as standalone Gazebo plugins such as the gazebo differential drive plugin. 

Thanks for pointing out those posts, the comments and changes of the authors would be very much valued for fixing our tutorials. Using open-source and contributing back are essential for keeping things running. We can't go out hunting for posts telling if a tutorial is wrong, we are to be approached.

The mode-switch should also be implemented for the Gazebo `RobotHW` so any joint interface could be used for the example robots. I personally can't allocate time to that now but would be happy to review PR-s on it.

JenniferBuehler, 2017-11-29:
Thanks for letting me know about this. I was initially planning to make a PR proposing the addition of velocity controllers like the one I was playing around with, but haven't found the time yet. My package is now over a year old (or almost 2, oh time flies!) so it probably would need updating again anyway. If you think it would be really useful to have the sample velocity controller PR soon, I'll try to move this up in my TODO list :)

jacknlliu, 2017-11-30:
hope you share your research about this up to date

destogl, 2017-12-05:
Hi,

we have some solution for generic Cartesian controllers extending ros_control controllers and using moveIt IK-plugin interface to dynamically load different IKs. We tested them on multiple ros_control enabled robots and they work perfectly. Would this be of your interests?

jacknlliu, 2017-12-05:
:thumbsup: could you share your codes about these? And is it tested with Gazebo RobotHW?