ros-controls/ros_controllers #305: four_wheel_steering_controller orientation test

bmagyar, 2017-11-28:
Hi @vincentrou ,

Could you please have a look at the orientation test?

https://travis-ci.org/ros-controls/ros_controllers/jobs/308425784

vincentrou, 2017-11-28:
It seems that the orientation tolerance is too low.
`const double ORIENTATION_TOLERANCE = 0.03; // 0.57 degree precision`
I will increase it.

bmagyar, 2017-11-28:
Be generous, I think I saw a bit of fluctuation on the precision.

bmagyar, 2017-12-05:
Hi @vincentrou ,

It is still failing... 
https://travis-ci.org/ros-controls/ros_controllers/jobs/305343529

Also I think that it is better to use smaller hardcoded sleeps, that helps getting our travis jobs through in a timely manner. Can you try a faster movement with a shorter time?

vincentrou, 2017-12-07:
I don't see errors on the build that you pointed.

bmagyar, 2017-12-19:
Sorry. I should have copied the text, it seems re-running travis jobs puts things under the same hash still...

The idea is still the same: could you change the tests a bit to last for less time? Let's see how those work (this will also trigger some new jobs so we see more failures).

bmagyar, 2017-12-19:
Got one today:

```
[ROSTEST]-----------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testForward][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testCrab][FAILURE]

/root/catkin_ws/src/ros_controllers/four_wheel_steering_controller/test/src/four_wheel_steering_4ws_cmd_test.cpp:84

The difference between sqrt(dx*dx + dy*dy) and cmd_vel.speed*travel_time is 0.051321167564745362, which exceeds POSITION_TOLERANCE, where

sqrt(dx*dx + dy*dy) evaluates to 1.5486788562771125,

cmd_vel.speed*travel_time evaluates to 1.6000000238418579, and

POSITION_TOLERANCE evaluates to 0.01.

--------------------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testCrab][FAILURE]

/root/catkin_ws/src/ros_controllers/four_wheel_steering_controller/test/src/four_wheel_steering_4ws_cmd_test.cpp:85

The difference between dx and cmd_vel.speed*travel_time*cos(cmd_vel.front_steering_angle) is 0.047414576083030235, which exceeds POSITION_TOLERANCE, where

dx evaluates to 1.430792691270983,

cmd_vel.speed*travel_time*cos(cmd_vel.front_steering_angle) evaluates to 1.4782072673540132, and

POSITION_TOLERANCE evaluates to 0.01.

--------------------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testCrab][FAILURE]

/root/catkin_ws/src/ros_controllers/four_wheel_steering_controller/test/src/four_wheel_steering_4ws_cmd_test.cpp:86

The difference between dy and cmd_vel.speed*travel_time*sin(cmd_vel.front_steering_angle) is 0.019639761074799367, which exceeds POSITION_TOLERANCE, where

dy evaluates to 0.59265375598685188,

cmd_vel.speed*travel_time*sin(cmd_vel.front_steering_angle) evaluates to 0.61229351706165125, and

POSITION_TOLERANCE evaluates to 0.01.

--------------------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testSymmetricTurn][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testNonSymmetricTurn][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testOdomFrame][passed]

SUMMARY

 * RESULT: FAIL

 * TESTS: 5

 * ERRORS: 0

 * FAILURES: 1
```

bmagyar, 2017-12-19:
```
[ROSTEST]-----------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testForward][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testCrab][FAILURE]

/root/catkin_ws/src/ros_controllers/four_wheel_steering_controller/test/src/four_wheel_steering_4ws_cmd_test.cpp:84

The difference between sqrt(dx*dx + dy*dy) and cmd_vel.speed*travel_time is 0.011993394578713401, which exceeds POSITION_TOLERANCE, where

sqrt(dx*dx + dy*dy) evaluates to 1.5880066292631445,

cmd_vel.speed*travel_time evaluates to 1.6000000238418579, and

POSITION_TOLERANCE evaluates to 0.01.

--------------------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testCrab][FAILURE]

/root/catkin_ws/src/ros_controllers/four_wheel_steering_controller/test/src/four_wheel_steering_4ws_cmd_test.cpp:85

The difference between dx and cmd_vel.speed*travel_time*cos(cmd_vel.front_steering_angle) is 0.011080451726449558, which exceeds POSITION_TOLERANCE, where

cmd_vel.speed*travel_time*cos(cmd_vel.front_steering_angle) evaluates to 1.4782072673540132, and

POSITION_TOLERANCE evaluates to 0.01.

--------------------------------------------------------------------------------

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testSymmetricTurn][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testNonSymmetricTurn][passed]

[four_wheel_steering_controller.rosunit-four_wheel_steering_controller_4ws_cmd_test/testOdomFrame][passed]

SUMMARY

 * RESULT: FAIL

 * TESTS: 5

 * ERRORS: 0

 * FAILURES: 1
```

commit 42d0410b (2017-11-30): Merge pull request #306 from vincentrou/fix_orientation_test

[4ws_controller] fix #305 Increase orientation tolerance
commit 5616d203 (2017-11-30): [4ws_controller] fix #305 Increase orientation tolerance
PR #309 (2018-01-11): [4ws] Increase position tolerance
fix  #305
closed 2018-01-11 by bmagyar