ros-controls/ros_controllers #307: The parameter format of velocity_controllers in control.yaml

GianMeng, 2017-12-14:
Hi,I am new here.So I really don't know @ who to ask for help. **## _My problem is when I am using the controllers of velocity_controllers in .yaml file, I can find little information about  how to use them._** At first, I want to find some examples like my robot(The wheel only need to roll to decide go ahead or draw back, and the rotation axises above the wheels only need to rotate to decide turn left or turn right. Of course, it need to stop before it turn left or right.), but no example can be found. So I need to write it by myself. But when I trying to using the controllers of velocity_controllers and position_controllers, I can find little information about  how to use them. Unlike diff_driver_controllers, its controllers' parameters are given,such as max velocity and so on. So could you please tell me the parameter format of those controllers. And I would appreciate it  if you can give me some advise about what controllers my robot need to use. Thank you so much. 
[image]

GianMeng, 2017-12-14:
@trainman419

GianMeng, 2017-12-15:
@wjwwood

GianMeng, 2017-12-15:
@bmagyar

miguelprada, 2017-12-15:
Hi @GianMeng. First of all, please avoid pinging people like you did so shortly after opening the issue. Some may consider it a bit rude.

At a quick glance, it looks like the robot you're trying to control is a perfect match for the recently added [four_wheel_steering_controller](https://github.com/ros-controls/ros_controllers/tree/kinetic-devel/four_wheel_steering_controller). Can you have a look at it and see whether it fits your needs? Having a look at the [tests](https://github.com/ros-controls/ros_controllers/tree/kinetic-devel/four_wheel_steering_controller/test) can maybe help you figure out how to properly configure it.

If that does not work for you, you can probably deduce the settings required by the controllers in `velocity_controllers` by looking at the sources for [`forward_command_controller`](https://github.com/ros-controls/ros_controllers/tree/kinetic-devel/forward_command_controller/include/forward_command_controller).

GianMeng, 2017-12-16:
Oh, I'm so sorry for pinging people in a rude way. And @miguelprada thank you so much for answering my questions. I wil try to ues the controller and if it's ok, I wil report it here. Sorry and thank you so much again.

closed 2018-03-23 by bmagyar