ros-controls/ros_controllers #308: JointStateController doesn't get the effort with effort_controllers/effort_joint_interface

scnsh, 2017-12-21:
Hi, I found that JointStateController always doesn't get any efforts with **effort_controllers/effort_joint_interface**. 
But, JointStateController can get efforts with **position_controllers/effort_joint_interface**.

[image]

If someone knows the reason, let me know.
Thanks in advance.

bmagyar, 2017-12-21:
Could you specify it a bit more? 
What are you running? Simulation, real robot? Which ROS version, what operating system?
Also there's a single JointStateController which reads things from whatever is on the JointStateInterface. 
What controllers are you running? What are you expecting to see? Does the robot move in which case?

scnsh, 2017-12-21:
Hi, I use and edit pkg, gazebo_ros_demo(see https://github.com/ros-simulation/gazebo_ros_demos) to view rrbot in gazebo.
And, I edit following line on https://github.com/ros-simulation/gazebo_ros_demos/blob/kinetic-devel/rrbot_control/config/rrbot_control.yaml

from
> type:effort_controllers/JointPositionController

to

> type:effort_controllers/JointEffortController

I expect to get efforts from rostopic **/rrbot/joint_states** while rrbot is moving in Gazebo.
But efforts are always zero( in previous post).

PatrickHussey, 2018-01-19:
I can confirm this as well. I ported a simulation from indigo to gazebo that uses position_controllers/JointTrajectoryController for arm joints and effort_controllers/GripperActionController for a gripper and all are reporting zero effort in the /jointstate message regardless of the orientation or motion. 

To add another issue, it seems the effort_controllers/GripperActionController no longer works regardless of how the PID gains are set in the controller configuration and what the effort/velocity/position limits get set to in the URDF

All of this works fine in Indigo/Gazebo2

PatrickHussey, 2018-01-19:
Just to follow up
If I use effort_controllers/GripperActionController for the Gazebo controller type and specify hardware_interface/EffortJointInterface in the transmission nothing works.

If I switch to position_controllers/GripperActionController for the Gazebo controller type and specify hardware_interface/PositionJointInterface in the transmission everything works, but the joint states still all report 0 effort. So this is still an issue

bmagyar, 2018-01-20:
Thanks for the report guys.
Could you please redirect them to https://github.com/ros-
simulation/gazebo_ros_pkgs/tree/kinetic-devel/gazebo_ros_control ?
Values propagated to the joint interfaces is done by the specific RobotHW
implementation, it's not related to the controllers.
In this case your relevant RobotHW implementation is `gazebo_ros_control`.

PatrickHussey, 2018-01-21:
I think this has to do with the Gazebo API GetForce not being implemented after looking at the code. There is an old post here: https://answers.ros.org/question/9585/does-jointgetforce-return-the-sum-of-all-forces-on-the-joint/ that describes the issue. 

Thanks for the response, agreed original post should close here and open a new issue on gazebo_ros_control

scnsh, 2018-01-22:
Thanks to comments for this post, I deeply understand the relationship the control and gazebo_ros_control.
I close this post.

closed 2018-01-22 by scnsh