ros-controls/ros_controllers #311: How to use joint_trajectory_controller to smooth motion between multiple poses?

littlejohnyang, 2018-02-06:
I have been using joint_trajectory_controller with PosVelAccJointInterface on my 7 DOF arm implementation. However I am running into an issue with smoothing the motion. I noticed I can set velocity to zero so it triggers QuinticSplineSegment smooth function. It seems working fine if the trajectory contains only one pose (or trajectory point), and it smoothly accelerates and decelerates the joints. However, when my trajectory contains multiple trajectory points (setting velocity to zero for first and final trajectory point and a fixed velocity value for all in between points ), I noticed it does start and stop for every single trajectory execution. In other words, it keeps accelerating and decelerating multiple times other than just accelerating at the start of the trajectory and decelerating at the end of the trajectory.

What is the proper way to use this controller package to smoothly interpolate smooth motion between multiple trajectory input?

I noticed a related [question](https://answers.ros.org/question/244112/smooth-arm-trajectory-with-jointtrajectorypoint/) posted on ROS Answer has got no answer since last year, so I am hoping to find some help direction from the developer. Thanks.

cmaestre, 2018-03-14:
I have reproduced exactly the same behaviour in my experiment. Any ideas about how to fix this problem? Thanks

ipa-mdl, 2018-03-15:
Have you checked http://wiki.ros.org/joint_trajectory_controller#Trajectory_representation?

cmaestre, 2018-03-16:
I did check everything, yes. For N goals, I select a Cartesian 3D position and MoveIt transforms it into a set of positions, velocities and accelerations (using [computeCartesianPath](http://docs.ros.org/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#ad6b02d15000d5b17c89b15a0f744b47c)). This information is passed as a new goal to the Action Server, which runs it preemting the previous goal, if needed.

The selected positions are just in a straight line reducing coordinate X, i.e. moving the end-effector closer to the Baxter robot. The distance between 2 positions is of 10cm, and a new goal is created each 0.7 seconds (before reaching the current goal):

```python
init_pos_x = 0.64
displacement_x = -0.1
delay = 0.7
   
for i in range(5):    
    pub_trajectory_motion_topic(
                         pub,
                         [init_pos_x + i*displacement_x,-0.1,0],
                         'right')
    time.sleep(delay)
```

When a new goal starts velocity and acceleration profiles go to 0 (see attached files and [video](https://youtu.be/4tmjsKkC5J0)).
[accelerations.txt](https://github.com/ros-controls/ros_controllers/files/1818855/accelerations.txt)
[velocities.txt](https://github.com/ros-controls/ros_controllers/files/1818856/velocities.txt)

bmagyar, 2018-03-23:
may have to do with #297 ?

cmaestre, 2018-03-26:
Yes, it looks like it is directly related. Since I am currently working in Indigo, how could I avoid this problem?

bmagyar, 2018-03-26:
I'm tempted to cherry-pick #297 to indigo-devel... What do you think @ipa-mdl ?

ipa-mdl, 2018-03-26:
I don't see how #297 is related (it only affects the end of the trajectory).
These are quite big jumps.

@cmaestre: could you prepare a plot of the `controller_state`, please? I wonder if the controller actually generares these jumps or if they are in the input data.

Some general notes:
1 .The topic is not meant for live streaming of points, just for complete (longer!) trajectories
2. `computeCartesianPath` does not generate smooth trajectories at all
3. In the video it looks like there is problem with the starting state.