ros-controls/ros_controllers #312: No trajectory defined at current time

dogoepp, 2018-02-27:
Hi,

With our [hardware interface](https://github.com/resibots/dynamixel_control_hw/) for Dynamixel servo-motors, I recently started getting the following occasional error:

```
Unexpected error: No trajectory defined at current time. Please contact the package maintainer.
```

The controller we use is `position_controllers/JointTrajectoryController` from this project and commands are sent with the rqt plugin for trajectory control.

Despite this error, there is no visible issue with the motion of the robotic arm controlled with this software. Do you have a hint on where I should search for the cause of this error ?

bmagyar, 2018-03-13:
It would be nice to know 
* at what rate are you sending commands to the `joint_trajectory_controller`
* how the timing of those commands are: you may be sending commands with timestamps in the past
* you may have a clogged control loop: time, feedback and commands out of sync

ipa-mdl, 2018-03-15:
Do you receive any other log message?
If not, please try to set the log level to debug.

dogoepp, 2018-03-16:
When "Joint Trajectory Controller" is enabled, it sends messages on the topic `/dynamixel_controllers/omni_arm_controller/command` at an average rate of 10Hz. The content of the message is constant (excet for seq):

```
header: 
  seq: 229
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
joint_names: [arm_0_1, arm_1_2, arm_2_3, arm_3_4, arm_4_5]
points: 
  - 
    positions: [-0.03, -0.03, -0.0, -0.03, -0.0]
    velocities: []
    accelerations: []
    effort: []
    time_from_start: 
      secs: 1
      nsecs:         0
```

When I move one of the sliders, the values in `positions` change.

It actually might be a problem with the control loop. It is not able to keep with time constraints. I'm investigating this end of the issue. Beside the warning in time contraints, I have no other log messages, despite printing at debug level.

dogoepp, 2018-03-16:
Due to an other issue, we are now limited to about 5Hz for a 5 Dynamixel Pro arm. If I stick to this (very low) frequency, the loop stays within a quarter period of time deviation (0.05 s).

Yet, whenever I enable the rqt trajectory controller, these messages keep popping up.