ros-controls/ros_controllers #313: /joint_states return error data when robot collision with object in gazebo

jacknlliu, 2018-03-05:
After robot and an object have a collision, and if the joint trajectory controller still sends joint positions to the robot, the `joint_states` will return the error data, even  the gazebo world and robot were reset by service call, the `joint_states` still return error data, and the `joint_trajectory_controller/state` also return error data!

`/joint_states`:
```yaml
header: 
  seq: 14205682
  stamp: 
    secs: 71085
    nsecs: 101000000
  frame_id: ''
name: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
position: [-8.51584976541768, -22393.868370150947, -23210.903847882473, -23059.99195414233, 7.4320796024417355, 9.18637489720941]
velocity: [0.00021875832846863282, 0.00585327683488699, 0.003416905486500389, -0.0001427822974524656, 6.232113829809081e-05, 1.6280227906504893e-06]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```

bmagyar, 2018-03-05:
This is to be expected, since the GazeboRosControl is not reset by these changes:
https://github.com/ros-simulation/gazebo_ros_pkgs/tree/kinetic-devel/gazebo_ros_control

jacknlliu, 2018-03-05:
@bmagyar thanks for your quick reply. It means that the `joint_state_controller` and `joint_trajectory_controller` can't receive the real joint state from gazebo. Is this expected? And how could we reset this or provide real `/joint_states` after reset gazebo world?

Thanks in advance!

jacknlliu, 2018-03-06:
` /joint_states` can't return normal data from the current model state, even if the robot has been moved to another position! `/joint_states` is broken?

bmagyar, 2018-03-23:
I'm sorry but you are submitting a whole lot of issues without a clear description of each. They may all boil down to the same underlying issue that's happening. Try to identify what's really wrong first.

closed 2018-03-23 by bmagyar