ros-controls/ros_controllers #317: joint_state_controller provides different joint_states from the gazebo plugin joint_state_publisher

jacknlliu, 2018-03-14:
When using `joint_state_controller` with gazebo, `/joint_states` will provide wrong and large joint data when robot has heavy collision with some object in the gazebo world. But if we use gazebo plugin `joint_state_publisher`, the joint states will be right! 

I find that gazebo plugin `joint_state_publisher` gets the joint states directly from gazebo, see [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_plugins/src/gazebo_ros_joint_state_publisher.cpp#L140). But I can't get where the joint states update in `joint_sate_controller`.

How does the `joint_state_[i]` update in the following codes?
https://github.com/ros-controls/ros_controllers/blob/e5a6c5a5f9f84756b6601961a5b03b7dffc5118c/joint_state_controller/src/joint_state_controller.cpp#L93

bmagyar, 2018-03-23:
It does it [exactly the same way](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros_control/src/default_robot_hw_sim.cpp#L257). My only guess is that you restart gazebo time.

jacknlliu, 2018-03-24:
It can't provide the exact joint data, even though we never reset gazebo time.

jacknlliu, 2018-03-29:
I found that the joint data from the gazebo API directly seems reasonable, see [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/kinetic-devel/gazebo_ros_control/src/default_robot_hw_sim.cpp#L257:23), but after this function [`angles::shortest_angular_distance()`](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/3164e4c6299407209a15fd87300f4364dc2e8063/gazebo_ros_control/src/default_robot_hw_sim.cpp#L265), the joint angles become very large after robot had a collision with some object in gazebo.

I think I should provide a basic ros package for proving this.

jacknlliu, 2018-09-21:
Hi, @bmagyar  Sorry for the delay. Could you try the [kinova_ros](https://github.com/Kinovarobotics/kinova-ros) package with the following command?
```sh
roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2n6s300
```
After the robot initial, we can find that `joint-state-controller` provides the wrong joint positions.

ignacioDeusto, 2022-07-05:
The issue has been detected and reported in the gazebo_ros_pkgs repository as the main bug concerns the gazebo_ros_control package sources. https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1405#issue-1292401140