ros-controls/ros_controllers #326: Pass effort through joint_trajectory_controller.

liarokapisv, 2018-03-30:
We would like to have a controller acting the same as position_controllers/JointTrajectoryController that would also forward effort values via a PosEff interface adaptor. The effort values do not need interpolation and could simply stay constant between TrajectoryPoints. Is this currently possible without rewriting a big part of the controller ? If not would it be possible to have a sketch of the required changes?

bmagyar, 2018-03-31:
I take it you are aware of how the hardware interfaces are [a template parameter](https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L64) in the controller implementation. Doesn't any of the current ones suit your needs?

I don't have the capacity to come up with new implementations at the moment but pull requests are welcome :)

liarokapisv, 2018-03-31:
Yes I am aware of that.  I believe that only changes the adaptor's behaviour when mapping the position/velocity/acceleration to the interface. The problem is that we need a PosVelEff interface and while writing a new adaptor for a new interface isn't hard in general, the controller apparently does not do anything with the trajectory message's effort values so I can't use them in the adaptor. What I propose is adding effort to the QuinticSplineSegment's state and simple do a linear interpolation, it is a natural addition to the controller's features. 

I can do the changes myself but I am not exactly sure of all the changes that need to happen to support this feature. On the top of my head I think the following changes are necessary:

Change QuanticSplineSegment to support effort sampling.
Change JointTrajectorySegment to read the messages' effort values.
Change initJointTrajectory
Appropriately change JointTrajectoryController's methods. 

The adaptor shouldn't need to change.

What do you think? Do you think this feature is useful for the generic trajectory controller?

amjack0, 2021-01-03:
Is there any update on this ?