ros-controls/ros_controllers #331: Release into Melodic

clalancette, 2018-04-10:
It looks like all of the dependencies for this package are available in Melodic, so it would be great to get it released there.  Thanks in advance.

bmagyar, 2018-04-17:
We have issues with one test that fails to build ATM: 
https://github.com/ros-controls/ros_control/issues/305#issuecomment-381353397

Also before releasing this I'd prefer to push through a shift to `std::shared_ptr` everywhere. Or would you prefer a release now (with the fixed test) and API change in 2 weeks?

clalancette, 2018-04-17:
@bmagyar The main thing this is holding up is `urdf_tutorial`, but I think that can wait another couple of weeks.  So let's hold off until you have the API change done.

clalancette, 2018-06-20:
This was done, so closing it out.  Thanks!

closed 2018-06-20 by clalancette