ros-controls/ros_controllers #332: gripper_action_controller pkg: gripper_cmd action client not connected

naveen010, 2018-04-11:
I am trying to add a gripper with 1dof joint to an arm and run it with Moveit. Below are the changes i made and when run the robot in gazebo, i´m getting this error.
[image]

Changes I made:
1. After i cloned and built the gripper_action_controller  package, the only change i made in the package is the gripper joint name.
2. To add to moveit, as suggested in #302 , in moveit config yaml file for controllers i added:
  - name: gripper_controller
    action_ns: gripper_cmd
    type: GripperCommand
    default: true
    joints:
      - gripper_joint

3. and to the controller configuration yaml file to launch robot in gazebo, for gripper controller, i added the below code which is given as an example configuration in one of the files in the gripper_action_controller package itself:
  gripper_controller:
  type: position_controllers/GripperActionController
  joint: gripper_joint
  goal_tolerance: 0.01
  stalled_velocity_threshold: 0.01
  stall_timeout: 0.2
  gains:
    gripper_joint: {p: 200, d: 1, i: 5, i_clamp: 1} 
and launched in the default_controllers calling the above yaml file.

I am not sure where exactly is the source of the error that causes the action client not to connect. Could i be missing something ? Do i need to make more changes either in the gripper_ or other packages ?. I am new to this stuff and really need help.

bmagyar, 2018-04-11:
>    1. and to the controller configuration yaml file to launch robot in

Are you sure the controller is running?
`rosservice call /controller_manager/list_controllers` should tell you. If
not, you should start it up first. Look for `spawner`.

naveen010, 2018-04-11:
I cross-checked using rosservice, the gripper_controller is not running. I manually loaded and started it, but when i run the rosservice list for controllers again, i get:

name: gripper_controller
    state: stopped
    type: position_controllers/GripperActionController
    claimed_resources: []
It looks like state is going back to stop as soon as it is started; and there is no hardware interface and resources shown, as it shows for my arm_controller:

name: arm_controller
    state: running
    type: position_controllers/JointTrajectoryController
    claimed_resources: 
      - 
        hardware_interface: hardware_interface::PositionJointInterface
        resources: ['arm_1_joint', 'arm_2_joint', 'arm_3_joint', 'arm_4_joint', 'arm_5_joint', 'arm_6_joint', 'arm_7_joint']
 I have no clue why its behaving like that.

bmagyar, 2018-04-11:
You mentioned you changed the joint name in the configuration file for the
controller.
Is this an existing joint? Is the name correct?  Is gazebo_ros_control
aware of this joint?

bmagyar, 2018-04-13:
MimicJointPlugin prints come the [`MimicJointPlugin`](https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins/blob/master/src/mimic_joint_plugin.cpp#L141) 

Having loaded `gazebo_ros_control` doesn't mean you have all joints configured right. It is hard to guess what the problems are with your configuration based on snippets of logs.

My next guess is: this new joint you added lacks transmissions of the right type. The configuration above will look for a position interface on that joint.

In general: please don't clone the repository and compile code only to change a yaml file to support your joint name for the gripper. You can copy the config file over to your configuration package. Makes life simpler.

bmagyar, 2018-04-14:
We are using issues to track software-related issues of the software.
Please raise your question on [ROS answers](https://answers.ros.org/questions/)

closed 2018-04-14 by bmagyar