ros-controls/ros_controllers #348: joint_trajectory_controller error when catkin build

dvof, 2018-05-25:
error: ‘ControllerBaseSharedPtr’ in namespace ‘controller_interface’ does not name a type

Two warnings, four errors. All errors are the same. Beneath all the output is shown.  

`In file included from /home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/robot_hw.h:35:0,
                 from /home/daan/tiago_test_ws/src/ros_control/controller_interface/include/controller_interface/controller_base.h:37,
                 from /opt/ros/kinetic/include/controller_manager/controller_spec.h:41,
                 from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot.cpp:36:
/home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h: In static member function ‘static std::vector<std::__cxx11::basic_string<char> > hardware_interface::CheckIsResourceManager<T>::callGR(T*, ...)’:
/home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:95:66: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static std::vector<std::string> callGR(T* iface, ...) { return {}; }
                                                                  ^
In file included from /home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/robot_hw.h:35:0,
                 from /home/daan/tiago_test_ws/src/ros_control/controller_interface/include/controller_interface/controller_base.h:37,
                 from /opt/ros/kinetic/include/controller_manager/controller_spec.h:41,
                 from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot_wrapping.cpp:38:
/home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h: In static member function ‘static std::vector<std::__cxx11::basic_string<char> > hardware_interface::CheckIsResourceManager<T>::callGR(T*, ...)’:
/home/daan/tiago_test_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/interface_manager.h:95:66: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static std::vector<std::string> callGR(T* iface, ...) { return {}; }
                                                                  ^
In file included from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36:0,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot.cpp:36:
/opt/ros/kinetic/include/controller_manager/controller_spec.h: At global scope:
/opt/ros/kinetic/include/controller_manager/controller_spec.h:56:25: error: ‘ControllerBaseSharedPtr’ in namespace ‘controller_interface’ does not name a type
   controller_interface::ControllerBaseSharedPtr c;
                         ^
In file included from /opt/ros/kinetic/include/controller_manager/controller_manager.h:36:0,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot_wrapping.cpp:38:
/opt/ros/kinetic/include/controller_manager/controller_spec.h: At global scope:
/opt/ros/kinetic/include/controller_manager/controller_spec.h:56:25: error: ‘ControllerBaseSharedPtr’ in namespace ‘controller_interface’ does not name a type
   controller_interface::ControllerBaseSharedPtr c;
                         ^
In file included from /opt/ros/kinetic/include/controller_manager/controller_manager.h:54:0,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot.cpp:36:
/opt/ros/kinetic/include/controller_manager/controller_loader_interface.h:49:33: error: ‘ControllerBaseSharedPtr’ in namespace ‘controller_interface’ does not name a type
   virtual controller_interface::ControllerBaseSharedPtr createInstance(const std::string& lookup_name) = 0;
                                 ^
In file included from /opt/ros/kinetic/include/controller_manager/controller_manager.h:54:0,
                 from /home/daan/tiago_test_ws/src/ros_controllers/joint_trajectory_controller/test/rrbot_wrapping.cpp:38:
/opt/ros/kinetic/include/controller_manager/controller_loader_interface.h:49:33: error: ‘ControllerBaseSharedPtr’ in namespace ‘controller_interface’ does not name a type
   virtual controller_interface::ControllerBaseSharedPtr createInstance(const std::string& lookup_name) = 0;
                                 ^
make[2]: *** [CMakeFiles/rrbot_wrapping.dir/test/rrbot_wrapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/rrbot_wrapping.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rrbot.dir/test/rrbot.cpp.o] Error 1
make[1]: *** [CMakeFiles/rrbot.dir/all] Error 2
make: *** [all] Error 2`

bmagyar, 2018-05-25:
That typedef was added [here](https://github.com/ros-controls/ros_control/commit/17b626ae04d9838b48e87e3e5dc5f6858276f07f#diff-4fd0994e27c30dd696cefc633db126a8R165) and has been released as per 0.13.3. Please don't mix and match different versions of `ros_control` and `ros_controllers`. I'd suggest to install `joint_trajectory_controller` and friends from packages, instead of building them from source.

bmagyar, 2018-05-25:
FYI @vlopez @jordi-pages you guys may want to revise the .rosinstall for Tiago.

closed 2018-05-25 by bmagyar

tobeperson, 2019-12-30:
> That typedef was added [here](https://github.com/ros-controls/ros_control/commit/17b626ae04d9838b48e87e3e5dc5f6858276f07f#diff-4fd0994e27c30dd696cefc633db126a8R165) and has been released as per [...]

Thank you very much!