ros-controls/ros_controllers #352: Building against the latest version of actionlib in Melodic fails

clalancette, 2018-06-20:
This is mostly a place to work with @bmagyar about the current problems between actionlib, ros_controllers, and realtime_tools.

## Background
In the leadup to Melodic, we ended up adding additional typedefs to actionlib to support ros_controllers: https://github.com/ros/actionlib/commit/506541aa5f0d23079f13f29cc8afa8f3e68fa7f0 .  However, it turned out that that caused build problems, so we quickly reverted that.  As it stands today, the versions in melodic of the packages involved here are:

* actionlib version 1.11.13-0 in melodic (one behind the latest release available on the indigo-devel branch)
* ros_controllers version 0.14.0-0 (the latest released on melodic-devel, but not the latest code)
* realtime_tools version 0.11.0-0 (one behind the latest release available on the melodic-devel branch)

## Current status
To try and move forward with this, I made a workspace containing actionlib, ros_controllers, and realtime_tools, and put them all on the latest code (indigo-devel for actionlib, melodic-devel for ros_controllers and realtime_tools).  When trying to build this workspace, I see the following error:

```
==> make -j1 in '/home/ubuntu/ros_ws/build_isolated/gripper_action_controller'
[  0%] Built target gripper_action_controller_xacro_generated_to_devel_space_
Scanning dependencies of target gripper_action_controller
[ 50%] Building CXX object CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o
In file included from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/home/ubuntu/ros_ws/install_isolated/include/realtime_tools/realtime_server_goal_handle.h:50:35: error: redeclaration of ‘typedef class std::shared_ptr<typename Action::_action_result_type::_result_type> realtime_tools::RealtimeServerGoalHandle<Action>::ResultPtr’
   typedef std::shared_ptr<Result> ResultPtr;
                                   ^~~~~~~~~
In file included from /home/ubuntu/ros_ws/install_isolated/include/actionlib/server/action_server.h:49:0,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:53,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/home/ubuntu/ros_ws/install_isolated/include/actionlib/action_definition.h:58:37: note: previous declaration ‘typedef class boost::shared_ptr<typename ActionSpec::_action_result_type::_result_type> realtime_tools::RealtimeServerGoalHandle<Action>::ResultPtr’
   typedef boost::shared_ptr<Result> ResultPtr; \
                                     ^
/home/ubuntu/ros_ws/install_isolated/include/actionlib/action_definition.h:58:37: note: in definition of macro ‘ACTION_DEFINITION’
   typedef boost::shared_ptr<Result> ResultPtr; \
                                     ^~~~~~~~~
In file included from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:56:0,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/home/ubuntu/ros_ws/install_isolated/include/realtime_tools/realtime_server_goal_handle.h:51:37: error: redeclaration of ‘typedef class std::shared_ptr<typename Action::_action_feedback_type::_feedback_type> realtime_tools::RealtimeServerGoalHandle<Action>::FeedbackPtr’
   typedef std::shared_ptr<Feedback> FeedbackPtr;
                                     ^~~~~~~~~~~
In file included from /home/ubuntu/ros_ws/install_isolated/include/actionlib/server/action_server.h:49:0,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:53,
                 from /home/ubuntu/ros_ws/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/home/ubuntu/ros_ws/install_isolated/include/actionlib/action_definition.h:63:39: note: previous declaration ‘typedef class boost::shared_ptr<typename ActionSpec::_action_feedback_type::_feedback_type> realtime_tools::RealtimeServerGoalHandle<Action>::FeedbackPtr’
   typedef boost::shared_ptr<Feedback> FeedbackPtr;
                                       ^
/home/ubuntu/ros_ws/install_isolated/include/actionlib/action_definition.h:63:39: note: in definition of macro ‘ACTION_DEFINITION’
   typedef boost::shared_ptr<Feedback> FeedbackPtr;
                                       ^~~~~~~~~~~
CMakeFiles/gripper_action_controller.dir/build.make:62: recipe for target 'CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o' failed
make[2]: *** [CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o] Error 1
CMakeFiles/Makefile2:1576: recipe for target 'CMakeFiles/gripper_action_controller.dir/all' failed
make[1]: *** [CMakeFiles/gripper_action_controller.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'gripper_action_controller': 
  Command '['/home/ubuntu/ros_ws/install_isolated/env.sh', 'make', '-j1']' returned non-zero exit status 2
```

I don't currently have time to look into this, but I think we should get it fixed.  @bmagyar , do you happen to have some time to look at this and try to resolve it?  Thanks.

ipa-mdl, 2018-06-20:
I would suggest to trigger a coordinated release with `actionlib` including the pointer types and `realtime_tools` without these extra types (https://github.com/ros-controls/realtime_tools/pull/27), within the same sync window.

bmagyar, 2018-06-20:
Hi @clalancette,
Thanks for chasing us up. How is the build now? :)

clalancette, 2018-06-20:
@ipa-mdl Absolutely, otherwise I think they are going to fail.

@bmagyar Thanks for the fix, it seems to compile for me now in a local workspace.  In the past when I tried to fix it by just removing the duplicate pointer types, things still didn't work for me, but I don't have the notes to see why that was.  I'm running a prerelease right now just to confirm that this is OK; I'll comment here again when it is finished.  Assuming all goes well, then a new release of `ros_controllers` and `realtime_tools` would be in order, at which point we can bump actionlib back to the latest and merge those two.  Sound like a plan?

bmagyar, 2018-06-20:
Absolutely! I've actually done the same tests prior to my first PR to
actonlib, I don't remember having to add other changes.

sloretz, 2019-05-28:
I think this can be fixed in `melodic`. The downstream typedefs are identical in https://github.com/ros-controls/realtime_tools/commit/b5c6cc9383a6855e33040e6d41bddcdc7b89282c which [is released into melodic](https://github.com/ros/rosdistro/blob/680852343e8daf674034b35f3f8f23f37704c6d6/melodic/distribution.yaml#L5505). The compiler error in the original report should be gone since ["Once declared, a typedef-name may only be redeclared to refer to the same type again"](https://en.cppreference.com/w/cpp/language/typedef).

1. Create `melodic-devel` branch in `ros/actionlib`
2. Revert ros/actionlib#113
3. Release actionlib into melodic from `melodic-devel` branch (this issue is fixed at this point)
4. Resubmit ros-controls/realtime_tools#27 since the typedefs are defined upstream
5. Release `realtime_tools` into melodic.

bmagyar, 2019-05-28:
> identical in ros-controls/realtime_tools@b5c6cc9

I too think it is fixable, at that time we tried rolling it out I think it
was causing a few regressions from packages also defining these aliases. It
may also potentially trigger massive rebuilds.

jhbruhn, 2020-01-16:
How is the state of this issue? I currently have the problem described and cannot access older versions of actionlib easily.

ipa-mdl, 2020-01-16:
>How is the state of this issue?

I think this issue is already is already solved and be closed.

> I currently have the problem described and cannot access older versions of actionlib easily.

Our CI  jobs don't indicate any problem. Which versions of `actionlib` and `ros_control` are you using?

jhbruhn, 2020-01-16:
I am trying to compile `joint_trajectory_controller 0.15.0` using `actionlib 1.12.0`. I am also having the same problem with `gripper_action_controller 0.15.0`.

ewillsbae, 2020-02-18:
I am also still experiencing an issue with this. Everything is on the proper branch (melodic-devel) but catkin_make keeps yielding the following errors (in all painful completion):
`==> Processing catkin package: 'gripper_action_controller'
==> Building with env: '/home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/devel_isolated/ackermann_steering_controller/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/build_isolated/gripper_action_controller'
==> make -j8 -l8 in '/home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/build_isolated/gripper_action_controller'
[  0%] Built target gripper_action_controller_xacro_generated_to_devel_space_
[ 50%] Building CXX object CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o
In file included from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:52:0,
                 from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:60:3: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   ResultPtr preallocated_result_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~
   Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:61:3: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
   FeedbackPtr preallocated_feedback_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~~~
   Feedback
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:50: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   RealtimeServerGoalHandle(GoalHandle &gh, const ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                  ^~~~~~~~~
                                                  Result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:63:115: error: ‘FeedbackPtr’ does not name a type; did you mean ‘Feedback’?
 st ResultPtr &preallocated_result = ResultPtr((Result*)NULL), const FeedbackPtr &preallocated_feedback = FeedbackPtr((Feedback*)NULL))
                                                                     ^~~~~~~~~~~
                                                                                                                   Feedback
In file included from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:52:0,
                 from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h: In constructor ‘realtime_tools::RealtimeServerGoalHandle<Action>::RealtimeServerGoalHandle(realtime_tools::RealtimeServerGoalHandle<Action>::GoalHandle&, const int&, const int&)’:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:68:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_result_’
       preallocated_result_(preallocated_result),
       ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:69:7: error: class ‘realtime_tools::RealtimeServerGoalHandle<Action>’ does not have any field named ‘preallocated_feedback_’
       preallocated_feedback_(preallocated_feedback)
       ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: error: ‘preallocated_result_’ was not declared in this scope
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:71:10: note: suggested alternative: ‘preallocated_result’
     if (!preallocated_result_)
          ^~~~~~~~~~~~~~~~~~~~
          preallocated_result
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: error: ‘preallocated_feedback_’ was not declared in this scope
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h:73:10: note: suggested alternative: ‘preallocated_feedback’
     if (!preallocated_feedback_)
          ^~~~~~~~~~~~~~~~~~~~~~
          preallocated_feedback
In file included from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/include/gripper_action_controller/gripper_action_controller.h:52:0,
                 from /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/src/ros_controllers/gripper_action_controller/src/gripper_action_controller.cpp:34:
/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h: In instantiation of ‘void gripper_action_controller::GripperActionController<HardwareInterface>::goalCB(gripper_action_controller::GripperActionController<HardwareInterface>::GoalHandle) [with HardwareInterface = hardware_interface::PositionJointInterface; gripper_action_controller::GripperActionController<HardwareInterface>::GoalHandle = actionlib::ServerGoalHandle<control_msgs::GripperCommandAction_<std::allocator<void> > >]’:
[... cut ...]
<== Failed to process package 'gripper_action_controller': 
  Command '['/home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/devel_isolated/ackermann_steering_controller/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/build_isolated/gripper_action_controller && /home/CYBERNET/ewills/gazebo_sims/alexnet_demo_py3/devel_isolated/ackermann_steering_controller/env.sh make -j8 -l8

Command failed, exiting.
`

bmagyar, 2020-02-18:
`realtime_tools` has the fix for this which should propagate to `ros_controllers`, otherwise we would see issues on travis and the ROS buildfarm. Don't worry, the problem is only with your setup :) 
Please read the error messages more carefully, it clearly tells you that `realtime_tools` is at fault and I suspect you all have an old version of it.

ewillsbae, 2020-02-18:
I'm on the branch `origin/melodic-devel` for `realtime_tools` and using the latest version of it. I agree it is probably my set up, but it's not an old version of `realtime_tools`, which is why I came here. It seemed the closest related error - I'm happy to be directed elsewhere.

bmagyar, 2020-02-18:
Your build is not using the local checkout of `realtime_tools`, but the system one.

> **/opt/ros/melodic/include/realtime_tools/realtime_server_goal_handle.h**:60:3: error: ‘ResultPtr’ does not name a type; did you mean ‘Result’?
   ResultPtr preallocated_result_;  // Preallocated so it can be used in realtime
   ^~~~~~~~~
   Result

ewillsbae, 2020-02-18:
Ok, thanks. I'm sorry I missed that. Please forgive a newer question, but how do I force it to do so? It seems to use the rest of the locally checked out repos.

ewillsbae, 2020-02-19:
In case this helps anyone else- it turns out that `git clone` of actionlib automatically goes to the branch indigo-devel no matter what you click on/go through. Switching to the melodic branch automatically fixes the issue.