ros-controls/ros_controllers #354: Suggest adding acceleration / braking limits instead of max and min acceleration limits for diff_drive_controller

shahvi, 2018-07-21:
In the diff_drive_controller, the current max / min acceleration limits work fine for setting limits for a forward moving robot.  However, these limits become reversed when going backwards. 
For large robots it would be good to have low acceleration limits so that the motion is smooth when starting to move.  However, for safety it would be good to still have high braking acceleration (deceleration) limits so that the robot can come to a quick stop.  Even when driving in reverse, it should have this same behavior. 
As per my understanding the current max / min acceleration specification doesn't allow this.

poorboy44, 2019-01-08:
I have the same problem. When driving backwards, I can't stop fast enough because of the low forward acceleration limits I have installed, to create smooth forward acceleration.

tomy983, 2022-07-30:
Would really love this too!

destogl, 2022-07-30:
This sounds great! Can you do a proposal/PR?