ros-controls/ros_controllers #358: joint_current_controller is needed

itfanr, 2018-09-20:
Some servo does not give a effort interface, but give a current interface instead.

So a joint_current_controller is needed which can send sin function to motor driver.

bmagyar, 2018-09-21:
Thank you! You are welcome to submit a pull request implementing this.

closed 2018-09-21 by bmagyar

itfanr, 2018-09-27:
@bmagyar  I am a beginner. But if I am glad to share my code.

ipa-mdl, 2018-09-27:
Would the implementation of a current-based controller be different from an effort-based controller?
In `ros_canopen` I have just used the effort interface for the current-based control interface.
`sensor_msgs/JointState` only supports effort anyway..

itfanr, 2018-09-28:
@ipa-mdl
I will read your [ros_canopen](https://github.com/ros-industrial/ros_canopen) repo and learn to control effort and current.

Thank you.