ros-controls/ros_controllers #359: [diff_drive_controller] Jerk Limits Behave Weirdly when Stopping Robot

rohbotics, 2018-09-23:
With a low jerk limit set the robot goes backward a bit after stopping. 

This is because the jerk limiter is making decisions by taking the 2nd derivative of velocity, so when the robot reaches the target of 0 the jerk limiter forces it to continue the negative acceleration by going backward. 

I would think the right behavior for the controller would be decreasing the acceleration as it approaches target velocity to limit the jerk. This would result in a v/t graph that looks like an s-curve.

rohbotics, 2018-09-23:
More info:
This is the expected (and actual) behavior with no jerk limits, and only acceleration limits. (infinite jerk)
[image]
(Red is Velocity, Black is Acceleration, and Blue is Jerk for all graphs) 

This is the behavior I would expect with jerk limits set. 
[image]

The actual behavior is much different, as the jerk limiter causes the velocity to go negative.

closed 2020-07-09 by rohbotics

roncapat, 2021-09-02:
@rohbotics How do you solved?