ros-controls/ros_controllers #361: Where are the source code of effort_controllers ?

itfanr, 2018-09-29:
I just found some code:
```
#include <effort_controllers/joint_effort_controller.h>
#include <pluginlib/class_list_macros.hpp>

template <class T>
void forward_command_controller::ForwardCommandController<T>::starting(const ros::Time& time)
{
  // Start controller with 0.0 effort
  command_buffer_.writeFromNonRT(0.0);
}

PLUGINLIB_EXPORT_CLASS(effort_controllers::JointEffortController,controller_interface::ControllerBase)

```
But I want to read the source code and see what action server is started.

bmagyar, 2018-09-29:
Please have a look at the repository. Also, for questions like this please refer to ROS answers. This is an issue tracking platform, not meant for support.

closed 2018-09-29 by bmagyar

itfanr, 2018-09-30:
@bmagyar  Thank you.