ros-controls/ros_controllers #362: "freeze" in forward command controllers

wxmerkt, 2018-10-06:
Hi,
We have a common use case for safely stopping and restarting streaming forward command controllers - aka "freezing". We have previously added this in our robot hardware interface, however, the forward command controllers stayed agnostic to this and kept writing into the command buffer - which caused jerk issues when unfreezing (the applications will know). We also do not want this to be handled at the individual application level.
Thus, we'd like to move this up one level to the forward command controllers. I.e., have a service call that toggles a flag to either write the current state or the received command to the hardware handle.

Before we start replicating, do you have pointers to things that exist like this that we may have missed, common best practice, or things to consider?

Thanks :)

ipa-mdl, 2018-10-16:
Why don't you stop the controller?

bmagyar, 2018-10-24:
The only place where you can certainly stop commands making it through to your hardware is the hardware abstraction layer, RobotHW.

Your safety stop mode in the RobotHW then should know how to reset the state handles too.